Santroller/Santroller #171: [Request]: Button Function Switch

Proofeek, 2024-05-11:
### Please describe your request.

A setting that will change the button assignment to another. For example: Assign the second kick to change the function of the yellow cymbal to the blue cymbal. In this way, while the second kick is pressed, the yellow cymbal will trigger the blue one, and if the second kick is not pressed, it will trigger the yellow one.

I don't know how much it's needed or useful, but I thought that this way it would be possible to play pro charts in Rock Band on Guitar Hero drums, if the second kick is digital

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2024-05-11:
You can already do this sort of thing? If your talking about USB host inputs, you can just use "Add setting" and bind the setting yourself.

closed 2024-05-11 by sanjay900

Proofeek, 2024-05-11:
Santroller crashes when saving if select 'Analog Pin Input' in a shortcut

sanjay900, 2024-05-12:
Yeah I guess that makes sense, not really something I'll fix for a while
since it's not the most important issue