Santroller/Santroller #177: [Request]: Direct Number Input For Analog Value UIs

tloftis, 2024-05-22:
### Please describe your request.

It'd be nice to have a number input to go along with the Mix/Max/Deadzone sliders when playing with Analog values such as the drum or whammy bar.
Could have the input only appear when in advance mode, I find I want as close to the same Min/Max values for all drum faces to help be consistent.

Also a calibrate button for the drums/kick would be nice.

Thank you,
Tim

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2024-05-22:
Calibrate buttons for drums and kick don't really make as much sense as you'd think being that its trigger based

tloftis, 2024-05-22:
Yeah, I could see that. I was just hoping for something like a button to capture the highest value generated during a period of time. Like press that button to set Min then hitting around the Drumset without hitting the pad then press the button again to set the value. Then same with Max but just hitting the pad as if playing to set that value. Then with those numbers just slight tweaks if needed.

Really though, I'd just like to be able to see the number the slider has set for both Min and Max so I can match them and/or write them down and better tune values by specific increments.

I've completed a conversion a PS3 Guitar hero drumset using a Pico and analog multiplexer after a week or so of trial and error. Tuning it so the drums would register right without cross triggering or not triggering when wanted took quite a bit of time and tweaking (The Cymbals really like to cross talk with the pads). Works well now on my 360 both Wired and with your Bluetooth transceiver module.

sanjay900, 2024-05-22:
You did use the diodes right? that killed a lot of the cross talk for me

tloftis, 2024-05-23:
Yes, everything is actually working very well including the analog kick pedal. Made a PCB to mount all the resistors and diodes to with a carrier for the analog multiplexer, replaced all the Piezoelectric in the drums and cymbals and have them connect to the PCB with Mono Aux connectors for quick swapping and allowing me to connect the RB cymbals/any electric drum set to any pad/cymbal if I want to. Also added another Mono Aux port for a digital kick pedal like the rock band one. Really quite happy with how it turned out, thank you very much for creating this software allowing this to be possible.

It was just a lot of trial and error to calibrate the Min/Max zones, kind of just hitting trying to match what I though I seen the actual values hit. I have it calibrated pretty well now and it works great, I just think it would have been easier to see/set the numbers Min/Max where set to during that.

sanjay900, 2024-05-23:
Totally fair, direct number input is on my list i havent gotten around to it yet though

tloftis, 2024-05-23:
Awesome, I look forward to it and thank you.

I did some light work to figure out where the UI portion of code was to take a crack at trying my hand at it was but wasn't able to find it, nor was I able to figure out the format of the config file to try editing that directly.

sanjay900, 2024-05-23:
The config format is binary, and the UI isn't open source, hence why you couldn't find it.

sanjay900, 2024-06-09:
This is implemented in 10.1.118

closed 2024-06-09 by sanjay900