Santroller/Santroller #178: [Issue]: Rock Band effect pickup not working on wireless guitars

Moose1002, 2024-05-23:
### Please describe your issue.

When attempting to use the Rock Band effect pickup switch while it is connected to another pi pico in wireless mode (one in guitar as a transmitter and one as a receiver) the effect switch doesn't seem to do anything, the game just locks it to an effect, that you can't change with the pickup switch.

I have verified that the effect switch does work when plugged in via usb, the issue only arises when using wireless mode. Both the transmitter and receiver are programmed as Rock Band guitars.

### Version

Latest

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-05-23:
what console?

sanjay900, 2024-05-23:
v10.1.71 when it comes out should fix 360

Roet-Ivar, 2024-05-23:
I'm sorry for being lightly OT, but I'm very intrigued by the new wireless build. What average latency are we talking here compared to the 8ms of the wired connection?

sanjay900, 2024-05-23:
https://santroller.tangentmc.net/latency/

Roet-Ivar, 2024-05-23:
I appreciate your reply, although I don't think this was answering my question about wireless latency (it could also being bad at finding the information).

Moose1002, 2024-05-23:
> v10.1.71 when it comes out should fix 360

360 was correct, thanks for looking into it.

sanjay900, 2024-05-23:
https://santroller.tangentmc.net/latency/bluetooth.html

sanjay900, 2024-05-23:
Should be fixed now

closed 2024-05-23 by sanjay900