Santroller/Santroller #179: [Issue]: Set Guitar as Keyboard, but PC doesn't detect it

LilRenlor, 2024-05-24:
### Please describe your issue.

I mapped the guitar and saved, if I plug the guitar into my Android phone or Chromebook, it works, but on my PC, it doesn't work, no inputs.

### Version

10.1.74

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-05-24:
what are you trying to play?

sanjay900, 2024-05-24:
I suspect your using 6KRO, seems thats broken on windows but will be fixed with v10.1.77

If your goal is to play festival, instead of using Keyboard mode you can use normal guitar mode, and just hit `add setting` and add the Festival console mode

doing that will let it emulate a keyboard if you hold a button while plugging the controller in

sanjay900, 2024-05-24:
10.1.77 is out now

closed 2024-05-24 by sanjay900