Santroller/Santroller #182: [Issue]: PS2 adapter whammy bar can't be properly configured

noiob, 2024-05-26:
### Please describe your issue.

The whammy bar on my redoctane guitar can only go in one direction, but there doesn't seem to be a way to configure it to have it rest at 0 (it'll always rest at negative max)

### Version

10.1.96

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

PS2

sanjay900, 2024-05-26:
That is how xinput guitars work, xplorers do this as well.

closed 2024-05-26 by sanjay900

noiob, 2024-05-26:
I just don't wanna spin in place in fortnite

sanjay900, 2024-05-26:
https://santroller.tangentmc.net/tool/fortnite_festival.html

sanjay900, 2024-05-26:
We have a keyboard mode for festival, which means you don't need to use any mappers and fixes all these sorts of issues. works on console too.

noiob, 2024-05-27:
Since festival supports pro mode with any controller now I hadn't tried that specific mode, thank you!