Santroller/Santroller #184: [Issue]: Error when configuring controller

Grimicka, 2024-06-04:
### Please describe your issue.

Processing microdetect (platform: atmelavr; board: micro; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/micro.html
PLATFORM: Atmel AVR (5.0.0) > Arduino Micro
HARDWARE: ATMEGA32U4 16MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) External (simavr)
PACKAGES: 
 - framework-arduino-avr @ 5.1.0 
 - tool-avrdude @ 1.70200.0 (7.2.0) 
 - toolchain-atmelavr @ 3.70300.220127 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- LUFA
Building in release mode
"C:\Users\Grim\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages/tool-avrdude/avrdude" -C "C:\Users\Grim\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages/tool-avrdude/avrdude.conf" -p atmega32u4 -P COM3 -c avr109 -e
avrdude: AVR device initialized and ready to accept instructions
ignature = 0x1e9587 (probably m32u4)
avrdude: erasing chip
avrdude done.  Thank you.
========================= [SUCCESS] Took 4.48 seconds =========================
Processing sparkfun_promicro (board: sparkfun_promicro16; platform: atmelavr; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing felis/USB-Host-Shield-20 @ ^1.6.0
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: USB-Host-Shield-20@1.6.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
Uploading script to detect speed
Processing microdetect (platform: atmelavr; board: micro; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/micro.html
PLATFORM: Atmel AVR (5.0.0) > Arduino Micro
HARDWARE: ATMEGA32U4 16MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) External (simavr)
PACKAGES: 
 - framework-arduino-avr @ 5.1.0 
 - tool-avrdude @ 1.70200.0 (7.2.0) 
 - toolchain-atmelavr @ 3.70300.220127 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- LUFA
Building in release mode
Compiling .pio\build\microdetect\src\avr\lufa\LUFA.c.o
Compiling .pio\build\microdetect\src\avr\lufa\reboot.c.o
Compiling .pio\build\microdetect\src\avr\micro\detect\descriptors_detect.c.o
Compiling .pio\build\microdetect\src\avr\micro\detect\main.c.o
Compiling .pio\build\microdetect\src\avr\micro\shared\bootloader.c.o
Compiling .pio\build\microdetect\libcac\LUFA\LUFACDCSerial.cpp.o
Archiving .pio\build\microdetect\libFrameworkArduinoVariant.a
Compiling .pio\build\microdetect\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\microdetect\FrameworkArduino\HardwareSerial0.cpp.o
Compiling .pio\build\microdetect\FrameworkArduino\HardwareSerial1.cpp.o
Compiling .pio\build\microdetect\FrameworkArduino\HardwareSerial2.cpp.o
Compiling .pio\build\microdetect\FrameworkArduino\HardwareSerial3.cpp.o
Compiling .pio\build\microdetect\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\microdetect\FrameworkArduino\Print.cpp.o
Compiling .pio\build\microdetect\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\microdetect\FrameworkArduino\Tone.cpp.o
In file included from C:\Users\Grim\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages\framework-arduino-avr\libraries\LUFA/LUFA/LUFA/Drivers/USB/Core/ConfigDescriptors.h:56:0,
                 from C:\Users\Grim\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages\framework-arduino-avr\libraries\LUFA/LUFA/LUFA/Drivers/USB/USB.h:389,
                 from C:\Users\Grim\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages\framework-arduino-avr\libraries\LUFA/LUFA.c.inc:43,
[... 133 lines cut ...]
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://registry.platformio.org/search?q=header:Wire.h
*
**************************************************************
 #include <Wire.h>
          ^~~~~~~~
compilation terminated.
*** [.pio\build\sparkfun_promicro\src\shared\main\wii_emu.cpp.o] Error 1
========================= [FAILED] Took 28.04 seconds =========================

### Version

10.1.108

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

Grimicka, 2024-06-04:
Also, I had this working a few months ago, but changing it to the setting for Fortnite festival started it

sanjay900, 2024-06-05:
fixed with v10.1.110

closed 2024-06-05 by sanjay900