Santroller/Santroller #186: [Issue]: 'LED_DEBOUNCE_COUNT' was not declared in this scope

SwiggityCricket, 2024-06-09:
### Please describe your issue.

When programming a RPi Pico using the latest version of the windows (installed) configurator, the compiler can't seem to find "LED_DEBOUNCE_COUNT". See the error here:
[error.txt](https://github.com/user-attachments/files/15753282/error.txt)

### Version

v10.1.118

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

SwiggityCricket, 2024-06-09:
I manually edited shared_main.cpp and added `#define LED_DEBOUNCE_COUNT 1` and it compiled just fine. Not sure what the value is supposed to be though.

sanjay900, 2024-06-09:
Yeah, I'll fix this though I suspect it's only an issue for initial
programming

sanjay900, 2024-06-09:
fixed with v10.1.119

closed 2024-06-09 by sanjay900