Santroller/Santroller #187: [Issue]: Error while trying do save settings

xhirozzzz, 2024-06-13:
### Please describe your issue.

I'm changind the collors from my guitar, and the applications craches

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/60d6ae81fcc73c34b1493ca9e261695e471bc0c2.zip@1.11.0; board: pico; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.11.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.30200.0 (3.2.0) 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-rp2040-earlephilhower @ 5.120300.240125 (12.3.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 50 compatible libraries
Scanning dependencies...
Dependency Graph
|-- USB-Host-Shield-20 @ 1.6.2
|-- MIDI Library @ 5.0.2
|-- SPI @ 1.0
|-- Wire @ 1.0
|-- fxpt_math
Building in release mode
Compiling .pio\build\pico\src\pico\io.cpp.o
Compiling .pio\build\pico\src\pico\main.cpp.o
Compiling .pio\build\pico\src\pico\pins.cpp.o
Compiling .pio\build\pico\src\pico\xinput_device.cpp.o
Compiling .pio\build\pico\src\pico\xinput_host.cpp.o
Compiling .pio\build\pico\src\shared\hid\hid.cpp.o
Compiling .pio\build\pico\src\shared\main\adxl345.cpp.o
Compiling .pio\build\pico\src\shared\main\i2c_shared.cpp.o
Compiling .pio\build\pico\src\shared\main\max1704x.cpp.o
Compiling .pio\build\pico\src\shared\main\mpr121.cpp.o
Compiling .pio\build\pico\src\shared\main\ps2.cpp.o
Compiling .pio\build\pico\src\shared\main\ps2_emu.cpp.o
In file included from C:\Users\hirog\Documents\Clone Hero\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base/include/pico/assert.h:14,
                 from C:\Users\hirog\Documents\Clone Hero\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base/include/pico/types.h:12,
                 from C:\Users\hirog\Documents\Clone Hero\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base/include/pico.h:24,
                 from C:\Users\hirog\Documents\Clone Hero\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/rp2_common/hardware_adc/include/hardware/adc.h:10,
                 from src\pico\pins.cpp:1:
c:\users\hirog\documents\clone hero\santrollerconfigurator-win-portable\current\binaries\platformio\packages\toolchain-rp2040-earlephilhower\arm-none-eabi\include\c++\12.3.0\cassert:43:10: fatal error: bits/c++config.h: No such file or directory
   43 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from C:\Users\hirog\Documents\Clone Hero\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base/include/pico/assert.h:14,
                 from C:\Users\hirog\Documents\Clone Hero\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base/include/pico/types.h:12,
                 from C:\Users\hirog\Documents\Clone Hero\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base/include/pico.h:24,
                 from C:\Users\hirog\Documents\Clone Hero\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/rp2_common/hardware_spi/include/hardware/spi.h:10,
                 from src\pico/io_define.h:2,
                 from include/io.h:4,
                 from src\pico\io.cpp:1:
c:\users\hirog\documents\clone hero\santrollerconfigurator-win-portable\current\binaries\platformio\packages\toolchain-rp2040-earlephilhower\arm-none-eabi\include\c++\12.3.0\cassert:43:10: fatal error: bits/c++config.h: No such file or directory
   43 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from C:\Users\hirog\Documents\Clone Hero\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/tusb_option.h:198,
                 from src\pico\xinput_device.cpp:1:
C:\Users\hirog\Documents\Clone Hero\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/common/tusb_mcu.h:315: warning: "TUP_DCD_ENDPOINT_MAX" redefined
  315 |   #define TUP_DCD_ENDPOINT_MAX    16
      | 
In file included from C:\Users\hirog\Documents\Clone Hero\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/tusb_option.h:195:
[... 122 lines cut ...]
                 from .pio\libdeps\pico\USB-Host-Shield-20/settings.h:100,
                 from .pio\libdeps\pico\USB-Host-Shield-20/Usb.h:34,
                 from src\shared\hid\hid.cpp:6:
c:\users\hirog\documents\clone hero\santrollerconfigurator-win-portable\current\binaries\platformio\packages\toolchain-rp2040-earlephilhower\arm-none-eabi\include\c++\12.3.0\cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory
   41 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\pico\src\shared\hid\hid.cpp.o] Error 1
========================== [FAILED] Took 2.07 seconds ==========================

### Version

10.0.22631.0

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-06-14:
try the portable version, this just seems to happen for some people and i dont know why

sanjay900, 2024-06-16:
Ah, seems there is a problem when you put it in a directory with a space.
```c:\users\hirog\documents\clone hero\santrollerconfigurator-win-portable```
Try putting it somewhere where there isn't a space in one of the parent directories

xhirozzzz, 2024-06-17:
You are right, man. It worked when I redownloaded and launched on downloads path. But now for some reason the installed version is working just fine. Thanks for the support. If something goes wrong I’ll let you know.

closed 2024-06-17 by sanjay900