Santroller/Santroller #190: [Issue]: Error programming pico

CanOfPepsii, 2024-06-25:
### Please describe your issue.

Fails to write to the microcontroller

Log:
Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/60d6ae81fcc73c34b1493ca9e261695e471bc0c2.zip@1.11.0; board: pico; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing https://github.com/Santroller/USB_Host_Shield_2.0/archive/refs/heads/master.zip @ 1.6.2
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: USB-Host-Shield-20@1.6.2 has been installed!
Library Manager: Installing fortyseveneffects/MIDI Library @ 5.0.2
Downloading 0% 10% 20% 30% 40%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: MIDI Library@5.0.2 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.11.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.30200.0 (3.2.0) 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-rp2040-earlephilhower @ 5.120300.240125 (12.3.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 50 compatible libraries
Scanning dependencies...
Dependency Graph
|-- USB-Host-Shield-20 @ 1.6.2
|-- MIDI Library @ 5.0.2
|-- SPI @ 1.0
|-- Wire @ 1.0
|-- fxpt_math
Building in release mode
Compiling .pio\build\pico\FrameworkArduinoBootloader\boot2_w25q080_2_padded_checksum.S.o
Compiling .pio\build\pico\src\pico\bt.cpp.o
Compiling .pio\build\pico\src\pico\bt_rx.cpp.o
Compiling .pio\build\pico\src\pico\hidparser.c.o
Compiling .pio\build\pico\src\pico\io.cpp.o
Compiling .pio\build\pico\src\pico\main.cpp.o
Compiling .pio\build\pico\src\pico\midi_host.c.o
Compiling .pio\build\pico\src\pico\pins.cpp.o
Compiling .pio\build\pico\src\pico\xinput_device.cpp.o
Compiling .pio\build\pico\src\pico\xinput_host.cpp.o
Compiling .pio\build\pico\src\shared\config\config.c.o
Compiling .pio\build\pico\src\shared\controllers\controllers.c.o
Compiling .pio\build\pico\src\shared\hid\hid.cpp.o
Compiling .pio\build\pico\src\shared\main\adxl345.cpp.o
Compiling .pio\build\pico\src\shared\main\i2c_shared.cpp.o
Compiling .pio\build\pico\src\shared\main\max1704x.cpp.o
Compiling .pio\build\pico\src\shared\main\mpr121.cpp.o
In file included from W:\Documents\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/tusb_option.h:198,
                 from src\pico\midi_host.c:27:
W:\Documents\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/common/tusb_mcu.h:315: warning: "TUP_DCD_ENDPOINT_MAX" redefined
  315 |   #define TUP_DCD_ENDPOINT_MAX    16
      | 
In file included from W:\Documents\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/tusb_option.h:195:
include/tusb_config.h:7: note: this is the location of the previous definition
    7 | #define TUP_DCD_ENDPOINT_MAX 8
      | 
In file included from W:\Documents\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/tusb_option.h:198,
                 from src\pico\xinput_device.cpp:1:
W:\Documents\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/common/tusb_mcu.h:315: warning: "TUP_DCD_ENDPOINT_MAX" redefined
[... 62 lines cut ...]
src\pico\xinput_host.cpp:185:38: error: 'struct USB_Host_Data_t' has no member named 'genericSlider'; did you mean 'genericAxisSlider'?
  185 |                                 out->genericSlider = GetAxis(item);
      |                                      ^~~~~~~~~~~~~
      |                                      genericAxisSlider
Compiling .pio\build\pico\src\shared\usb\xsm3\usbdsec.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\xsm3.c.o
Generating linkerscript C:\Users\GT-R\AppData\Roaming\SantrollerConfigurator\Santroller\.pio\build\pico/memmap_default.ld
*** [.pio\build\pico\src\pico\xinput_host.cpp.o] Error 1
========================== [FAILED] Took 8.37 seconds ==========================

### Version

v10.1.146

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-06-25:
Yeah there's a typo on the latest release, use the 145 or 144

CanOfPepsii, 2024-06-25:
No worries I've already went back to an older build for now. 👍

sanjay900, 2024-06-25:
Fix was just pushed out as v10.1.148

closed 2024-06-25 by sanjay900