Santroller/Santroller #191: [Issue]: Analog sticks in Switch console mode don't zero correctly

Sulfrix, 2024-06-27:
### Please describe your issue.

I set up a Pi Pico to use as a USB controller adapter for my Switch. Everything works fine except for the analog sticks. The Switch's calibration screen does not register the analog stick as being properly zeroed, as it displays a ● instead of a ＋. The switch does register a zeroed analog stick whenever i move it slightly down and to the right. This still happens when I unplug my controller from the USB Host input.

### Version

v10.1.151

### What operating system are you using?

Linux

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-07-24:
I'll try and borrow a switch when I get back from my holiday and I'll take a look

sanjay900, 2024-09-20:
Fixed with v10.1.200

closed 2024-09-20 by sanjay900