Santroller/Santroller #192: Would it be possible to make it work with Fortnite Festival on Nintendo Switch?

XxyZwW, 2024-06-27:
### Please describe your request.

I tried to configure a wii adapter for wii guitar, to make it work in Fortnite Festival in Pro mode, only 2 compatibility options appear: PC (NKRO) and PS4/PS5 (6KRO), I already tried both by pressing the assigned button for it to work I detected it as a keyboard in NS, but neither way worked, I also tried with the normal guitar mode, but the adapter is still not detected, is it possible to add PRO mode compatibility (emulated keyboard) on Nintendo Switch?

The version of Santroller that I used was: 10.1.150

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Wii

sanjay900, 2024-06-27:
Normal guitar mode is detected on switch. However, you need to assign buttons in game. Make sure you assign the controller to player 1.

Switch doesn't support keyboards.

closed 2024-06-27 by sanjay900

XxyZwW, 2024-06-28:
I'll try that way, thank you very much