Santroller/Santroller #193: [Request]: More xinput controller types

sonik-br, 2024-07-04:
### Please describe your request.

Would be possible to add more controller types for xinput?
Like Flight Stick, Arcade Stick and Wheel?
I can see it's definition [here](https://github.com/Santroller/Santroller/blob/36b54ad639053942f3023fb318380e55b6a016fd/include/defines.h#L11-L13) But I could not find them being used.

Also, does it apply only for xbox 360 mode or also xbox one mode?

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2024-07-04:
Xinput types are only for 360, that's why the definitions aren't used anywhere else, since it doesn't really make sense to only be doing things for 360 only

sonik-br, 2024-07-04:
Understood.

I'm looking into a way to emulate a x360 flight stick (some flight games supports it, like Ace Combat).
As a workaround, if I change it [here](https://github.com/Santroller/Santroller/blob/36b54ad639053942f3023fb318380e55b6a016fd/include/config.h#L96) would it work?

sanjay900, 2024-07-04:
You can try it, see what happens

sanjay900, 2024-07-04:
Of course the other problem is that things like wheels and flight sticks would need to be fully supported and properly mappable when i do eventually support them, but for now you can certainly just patch the gamepad subtype and get away with that i suspect.

The 360 is probably the only console where you can easily do that, every other console has totally bespoke mappings for different types of controllers, so we would have to go out of our way to implement things like that elsewhere.

sonik-br, 2024-07-04:
I'm going to try it out.

I assumed that those special devices have it's own mapping.
Not sure how this thing maps out...
[image]

sonik-br, 2024-08-08:
Tested it as a racing wheel device and can confirm that it does work!
A game running on a x360 console detected it as a racing wheel.
Next step is to try with the Flight Stick subtype. But I still need to get a compatible game.

closed 2026-11-09 by sanjay900