Santroller/Santroller #195: [Issue]: Sudden error when writing

skylightlmnc, 2024-07-07:
### Please describe your issue.

Currently in the process of building a Guitar Hero Live controller. For some reason I get an error when trying to write to the Pico. I'm not that sure anymore but I think when the guitar was still incomplete and I was just testing the few buttons I have already soldered it was working fine. I also have a different Guitar Hero controller which works perfectly fine.

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/60d6ae81fcc73c34b1493ca9e261695e471bc0c2.zip@1.11.0; board: pico; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing https://github.com/Santroller/USB_Host_Shield_2.0/archive/refs/heads/master.zip @ 1.6.2
Downloading...
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: USB-Host-Shield-20@1.6.2 has been installed!
Library Manager: Installing fortyseveneffects/MIDI Library @ 5.0.2
Downloading 0% 10% 20% 30% 40%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: MIDI Library@5.0.2 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.11.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.30200.0 (3.2.0) 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-rp2040-earlephilhower @ 5.120300.240125 (12.3.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 50 compatible libraries
Scanning dependencies...
Dependency Graph
|-- USB-Host-Shield-20 @ 1.6.2
|-- MIDI Library @ 5.0.2
|-- SPI @ 1.0
|-- Wire @ 1.0
|-- fxpt_math
Building in release mode
Compiling .pio\build\pico\FrameworkArduinoBootloader\boot2_w25q080_2_padded_checksum.S.o
Compiling .pio\build\pico\src\pico\bt.cpp.o
Compiling .pio\build\pico\src\pico\bt_rx.cpp.o
Compiling .pio\build\pico\src\pico\hidparser.c.o
Compiling .pio\build\pico\src\pico\io.cpp.o
Compiling .pio\build\pico\src\pico\main.cpp.o
Compiling .pio\build\pico\src\pico\midi_host.c.o
Compiling .pio\build\pico\src\pico\pins.cpp.o
Compiling .pio\build\pico\src\pico\xinput_device.cpp.o
Compiling .pio\build\pico\src\pico\xinput_host.cpp.o
Compiling .pio\build\pico\src\shared\config\config.c.o
Compiling .pio\build\pico\src\shared\controllers\controllers.c.o
Compiling .pio\build\pico\src\shared\hid\hid.cpp.o
In file included from E:\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/tusb_option.h:198,
                 from src\pico\midi_host.c:27:
E:\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/common/tusb_mcu.h:315: warning: "TUP_DCD_ENDPOINT_MAX" redefined
  315 |   #define TUP_DCD_ENDPOINT_MAX    16
      | 
In file included from E:\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/tusb_option.h:195:
include/tusb_config.h:7: note: this is the location of the previous definition
    7 | #define TUP_DCD_ENDPOINT_MAX 8
      | 
Compiling .pio\build\pico\src\shared\main\adxl345.cpp.o
Compiling .pio\build\pico\src\shared\main\i2c_shared.cpp.o
In file included from E:\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/tusb_option.h:198,
                 from src\pico\xinput_host.cpp:27:
E:\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/common/tusb_mcu.h:315: warning: "TUP_DCD_ENDPOINT_MAX" redefined
  315 |   #define TUP_DCD_ENDPOINT_MAX    16
      | 
In file included from E:\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/tusb_option.h:195:
include/tusb_config.h:7: note: this is the location of the previous definition
    7 | #define TUP_DCD_ENDPOINT_MAX 8
      | 
In file included from E:\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/tusb_option.h:198,
                 from src\pico\xinput_device.cpp:1:
E:\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/common/tusb_mcu.h:315: warning: "TUP_DCD_ENDPOINT_MAX" redefined
  315 |   #define TUP_DCD_ENDPOINT_MAX    16
      | 
In file included from E:\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/tusb_option.h:195:
include/tusb_config.h:7: note: this is the location of the previous definition
    7 | #define TUP_DCD_ENDPOINT_MAX 8
      | 
Compiling .pio\build\pico\src\shared\main\max1704x.cpp.o
Compiling .pio\build\pico\src\shared\main\mpr121.cpp.o
Compiling .pio\build\pico\src\shared\main\ps2.cpp.o
Compiling .pio\build\pico\src\shared\main\ps2_emu.cpp.o
Compiling .pio\build\pico\src\shared\main\shared_main.cpp.o
In file included from E:\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/tusb_option.h:198,
                 from E:\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/common/tusb_common.h:70,
                 from E:\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/tusb.h:37,
                 from src\pico\main.cpp:6:
E:\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/common/tusb_mcu.h:315: warning: "TUP_DCD_ENDPOINT_MAX" redefined
  315 |   #define TUP_DCD_ENDPOINT_MAX    16
      | 
In file included from E:\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/lib/tinyusb/src/tusb_option.h:195:
include/tusb_config.h:7: note: this is the location of the previous definition
    7 | #define TUP_DCD_ENDPOINT_MAX 8
      | 
Compiling .pio\build\pico\src\shared\main\wii.cpp.o
Compiling .pio\build\pico\src\shared\main\wii_emu.cpp.o
Compiling .pio\build\pico\src\shared\usb\descriptors.cpp.o
Compiling .pio\build\pico\src\shared\usb\string_descriptors.c.o
Compiling .pio\build\pico\src\shared\usb\xbox.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\excrypt_des.c.o
In file included from include/config.h:10,
                 from include/pin_funcs.h:2,
                 from include/shared_main.h:1,
                 from src\shared\main\shared_main.cpp:1:
src\shared\main\shared_main.cpp: In function 'uint8_t tick_inputs(void*, USB_LastReport_Data_t*, uint8_t)':
include/config_data.h:389:17: error: 'struct PCFortniteRockBandGuitar_Data_t' has no member named 'rightShoulder'; did you mean 'leftShoulder'?
  389 |         report->rightShoulder = true;\
      |                 ^~~~~~~~~~~~~
src\shared\main\shared_main.cpp:1958:9: note: in expansion of macro 'TICK_XBOX_ONE'
 1958 |         TICK_XBOX_ONE;
      |         ^~~~~~~~~~~~~
include/config_data.h:390:17: error: 'struct PCFortniteRockBandGuitar_Data_t' has no member named 'rightShoulder'; did you mean 'leftShoulder'?
  390 |         report->rightShoulder=true;\
      |                 ^~~~~~~~~~~~~
src\shared\main\shared_main.cpp:1958:9: note: in expansion of macro 'TICK_XBOX_ONE'
 1958 |         TICK_XBOX_ONE;
      |         ^~~~~~~~~~~~~
include/config_data.h:408:17: error: 'struct PCFortniteRockBandGuitar_Data_t' has no member named 'leftThumbClick'
  408 |         report->leftThumbClick = true;\
      |                 ^~~~~~~~~~~~~~
src\shared\main\shared_main.cpp:1958:9: note: in expansion of macro 'TICK_XBOX_ONE'
 1958 |         TICK_XBOX_ONE;
      |         ^~~~~~~~~~~~~
Compiling .pio\build\pico\src\shared\usb\xsm3\excrypt_parve.c.o
*** [.pio\build\pico\src\shared\main\shared_main.cpp.o] Error 1
========================= [FAILED] Took 34.10 seconds =========================

### Version

v10.1.152

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-07-08:
> Library Manager: ***@***.*** has been installed!
> Library Manager: MIDI ***@***.*** has been installed!

Looks like you have a fortnite console mode binding set up still, remove
that because the fortnite bindings don't support the 6th fret

skylightlmnc, 2024-07-08:
I'm not really sure what you mean with fortnite console binding. I'm assuming you're talking about the fortnite instrument mapper . How would I remove these bindings?

skylightlmnc, 2024-07-08:
> Looks like you have a fortnite console mode binding set up still, remove that because the fortnite bindings don't support the 6th fret

I'm not really sure what you mean with fortnite console mode binding. I'm assuming you're talking about the fortnite instrument mapper? How would I remove these bindings?

sanjay900, 2024-07-08:
Try v10.1.153 when it finishes building

closed 2024-07-08 by sanjay900