Santroller/Santroller #197: [Request]: WS2812 support on arbitrary pins

DragRedSim, 2024-07-09:
### Please describe your request.

I've just put together a USB adapter for my Riffmaster for the 360, and I used an [RP2040-Zero board](https://www.waveshare.com/wiki/RP2040-Zero) to do so in a small size. Given that it uses the RP2040, it works exactly the same as a traditional Pico. This board has a built-in WS2812 LED, and I'd like to be able to program the adapter to use this single LED. Unfortunately, it is internally routed to GPIO16, and when attempting to apply a WS2812 LED, it only lists a few pins which are all MOSI SPI hardware pins, which do not include this pin. I understand that the APA-series LEDs require one of the hardware SPI pins, but the WS2812, as far as I can tell, should be able to use any pin?

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

N/A

DragRedSim, 2024-07-09:
Having just read through the WS2812 page on the guide, I see that in fact it is using the SPI hardware in a non-traditional manner in order to drive the LEDs, and as such, support for arbitrary pin selection is currently not possible. Closing this for the time being.

closed 2024-07-09 by DragRedSim

sanjay900, 2024-07-09:
Yeah, usually people bit bang the leds but I didn't want to do that as I didn't want to spend processing power on that, and I sadly don't have spare PIO instructions so I can't use that either :(