Santroller/Santroller #199: [Issue]: Configurator crashes every time I write a bluetooth controller on pico

Naddis4618, 2024-07-21:
### Please describe your issue.

Every time I try to write a Bluetooth controller with a pico w, it will say finished then lock up for a while, unfreeze for a half second then refreeze over and over again. 

### Version

Latest (as far as I know)

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-07-22:
Can I see a photo of your Pico W

Naddis4618, 2024-07-22:
> Can I see a photo of your Pico W , and a link to where you brought it from

Absolutely! Here is the link, and I will attach pictures of the board
itself! Thank you so much for your help!

Just found this amazing item on AliExpress. Check it out! $5.89 | Raspberry
Pi Pico W Board TYPE-C RP2040 Dual-Core ARM Low-Power Microcomputers
High-Performance Cortex-M0+ Processor https://a.aliexpress.com/_mqUMIRK

Noah Addis

sanjay900, 2024-07-22:
It's a fake Pico W that uses the wrong type of wireless chip. Sadly we
can't support these easily

Naddis4618, 2024-07-22:
Ahhhhhh, that makes sense, I appreciate your help. I’ll order a legit pico
W and try again! Thank you again!

Noah Addis

sanjay900, 2024-07-22:
Just make sure the listing doesn't say it's ESP based and you should be fine, the legit Pico W has a Infineon CYW43439