Santroller/Santroller #201: Implement Software Auto-Flam for Drums

DarkRTA, 2024-07-23:
### Please describe your request.

Due to bugs in how Rock Band 3 handles inputs with Pro Drums, some combinations of timings of inputs need to be automatically flammed to avoid issues with inputs being outright dropped by the game.

Currently trying to learn about all cases where this is nessecary, but the biggest one (from a righty perspective) is green pad + green cymbal which is special cased in lefty mode by the engine.

### What operating system are you using?

Linux

### What microcontroller are you making a request for?

N/A

### How will the microcontroller be programed for this request?

N/A

sanjay900, 2024-07-23:
Could probably tie the fix into how we handle debounce, where hitting both notes staggers the second one until after the debounce period

sanjay900, 2024-09-22:
in theory ive implemented flamming for any drum inputs that are hit at the same time in v10.1.216

closed 2024-09-22 by sanjay900