Santroller/Santroller #202: [Issue]: GH2 Whammy not working

patricksouza472, 2024-07-25:
### Please describe your issue.

I have configured my pico as a Santroller USB Adapter, I'm using my Rock Band guitar as a GH guitar and it works fine with GH3. GH2 partially works, the only thing missing is the Whammy bar support. RGH Xbox 360 with usbdsecpatch as nand patch. 

### Version

Latest

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

other issue not controller related or I have not gotten this far

sanjay900, 2024-07-25:
Did you calibrate the whammy within santroller

patricksouza472, 2024-07-25:
> Did you calibrate the whammy within santroller

Yes I did. Works perfectly on any other GH game (tested on GH3, GHWT and GH WoR)

sanjay900, 2024-07-25:
What about calibrating whammy in game (or is that only a thing on PS2)

patricksouza472, 2024-07-25:
> What about calibrating whammy in game (or is that only a thing on PS2)

Didn't try that, I didn't even know it was a thing on PS2 since I only played with the controllers back then. I'll try it and report back

patricksouza472, 2024-07-25:
> > What about calibrating whammy in game (or is that only a thing on PS2)
> 
> Didn't try that, I didn't even know it was a thing on PS2 since I only played with the controllers back then. I'll try it and report back

No dice, there's no option for it

sanjay900, 2024-08-24:
Have you got the guitar in rock band mode, or guitar hero mode?

sanjay900, 2024-12-23:
Do you mind trying this again with later versions, i wonder if this has changed at all

msuts, 2025-01-01:
I am on the latest version of Santroller and seeing the same issue with no GH2 whammy.  Whammy works in GH3 and presumably the other games as well. Haven't yet tested in RB1/2/3. The device is a Sparkfun Pro Micro 5V. All other functions work in GH2, just not whammy. Usbsecdpatch is running through DashLaunch.

sanjay900, 2025-01-02:
what does your calibration look like? seems to me that GH2 PS2 is just very very specific with what whammy values it likes, ive had reports that it works for some and doesn't for others

sanjay900, 2025-01-02:
Fixed with v10.1.315

closed 2025-01-02 by sanjay900