Santroller/Santroller #205: [Issue]: Midi Drums Don't Properly Trigger Tom Inputs

RobinatorX, 2024-08-02:
### Please describe your issue.

Every MIDI signal sent to the microcontroller outputs as its respective cymbal instead of the tom associated with it. For example, the yellow tom always activates the yellow cymbal. I've spent enough time pulling my hair out and trying different configurations that I'm not sure what to do now. The mode used is Rock Band Drums and the console type is PS3.

### Version

v10.1.153

### What operating system are you using?

Mac OS

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-08-19:
I suspect I know the cause of this, and will cut a new release when I get back from my holiday

sanjay900, 2024-08-24:
Should be fixed now

closed 2024-08-24 by sanjay900

RobinatorX, 2024-08-24:
You’re the best. Thanks for your service to this community🫡