Santroller/Santroller #206: [Issue]: Error when compiling PS2 Controller emulation or Wii Extension emulation

anabelgator, 2024-08-13:
### Please describe your issue.

Input: USB Host inputs
For PS2 controller emulation gives this error when compiling:

```
Compiling .pio\build\pico\src\shared\hid\hid.cpp.o
Compiling .pio\build\pico\src\shared\main\adxl345.cpp.o
Compiling .pio\build\pico\src\shared\main\max1704x.cpp.o
Compiling .pio\build\pico\src\shared\main\mpr121.cpp.o
Compiling .pio\build\pico\src\shared\main\ps2.cpp.o
Compiling .pio\build\pico\src\shared\main\ps2_emu.cpp.o
Compiling .pio\build\pico\src\shared\main\shared_main.cpp.o
Compiling .pio\build\pico\src\shared\main\wii.cpp.o
src\shared\main\ps2_emu.cpp: In function 'uint8_t receiveCommand()':
src\shared\main\ps2_emu.cpp:20:21: error: 'SPI_1_MISO' was not declared in this scope; did you mean 'SPI_0_MISO'?
   20 |     gpio_set_oeover(SPI_1_MISO, GPIO_OVERRIDE_LOW);
      |                     ^~~~~~~~~~
      |                     SPI_0_MISO
*** [.pio\build\pico\src\shared\main\ps2_emu.cpp.o] Error 1
```
For Wii Extension emulation:

```
Compiling .pio\build\pico\src\shared\hid\hid.cpp.o
Compiling .pio\build\pico\src\shared\main\adxl345.cpp.o
Compiling .pio\build\pico\src\shared\main\max1704x.cpp.o
Compiling .pio\build\pico\src\shared\main\mpr121.cpp.o
Compiling .pio\build\pico\src\shared\main\ps2.cpp.o
Compiling .pio\build\pico\src\shared\main\ps2_emu.cpp.o
Compiling .pio\build\pico\src\shared\main\shared_main.cpp.o
Compiling .pio\build\pico\src\shared\main\wii.cpp.o
Compiling .pio\build\pico\src\shared\main\wii_emu.cpp.o
In file included from include/config.h:10,
                 from include/pin_funcs.h:2,
                 from include/shared_main.h:1,
                 from src\shared\main\shared_main.cpp:1:
src\shared\main\shared_main.cpp: In function 'void tick_wiioutput()':
include/config_data.h:591:17: error: 'struct WiiClassicDataFormat3_t' has no member named 'leftThumbClick'
  591 |         report->leftThumbClick = true;\
      |                 ^~~~~~~~~~~~~~
src\shared\main\shared_main.cpp:1619:5: note: in expansion of macro 'TICK_WII'
 1619 |     TICK_WII;
      |     ^~~~~~~~
include/config_data.h:594:17: error: 'struct WiiClassicDataFormat3_t' has no member named 'rightThumbClick'
  594 |         report->rightThumbClick = true;\
      |                 ^~~~~~~~~~~~~~~
src\shared\main\shared_main.cpp:1619:5: note: in expansion of macro 'TICK_WII'
 1619 |     TICK_WII;
      |     ^~~~~~~~
Compiling .pio\build\pico\src\shared\usb\descriptors.cpp.o
*** [.pio\build\pico\src\shared\main\shared_main.cpp.o] Error 1
```

### Version

10.1.153

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

other issue not controller related or I have not gotten this far

sanjay900, 2024-08-13:
I'm probably going to rewrite the ps2 emulation when I get back from my holiday.

Interesting that the Wii emulation does that though, I assume that's only an issue for gamepad emulation and not for guitar or drums?

anabelgator, 2024-08-13:
> Interesting that the Wii emulation does that though, I assume that's only an issue for gamepad emulation and not for guitar or drums?

Yes, error occured on gamepad controller type

sanjay900, 2024-08-13:
I'll fix it when I'm back but for now you could probably get away with
disabling the mappings for left and right thumb click within the USB host
settings and that might let you use Wii emulation

anabelgator, 2024-08-14:
> I'll fix it when I'm back but for now you could probably get away with disabling the mappings for left and right thumb click within the USB host settings and that might let you use Wii emulation

Ok thank you.

Yes it works now when disabling left and right stick click on USB host mapping, and I just remembered that there is no thumbstick click on the actual wii classic controller 😄

sanjay900, 2024-08-24:
both of these should be fixed in the latest release when it finishes building

closed 2024-08-24 by sanjay900