Santroller/Santroller #207: [Issue]: When adding strum buttons, app crashes on config change

Remium599, 2024-08-13:
### Please describe your issue.

Everytime i save setting after adding strum up / down, the app crashes and i have to start over from scratch. 

Dump: 
Processing microdetect (platform: atmelavr; board: micro; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/micro.html
PLATFORM: Atmel AVR (5.0.0) > Arduino Micro
HARDWARE: ATMEGA32U4 16MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) External (simavr)
PACKAGES: 
 - framework-arduino-avr @ 5.1.0 
 - tool-avrdude @ 1.70200.0 (7.2.0) 
 - toolchain-atmelavr @ 3.70300.220127 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- LUFA
Building in release mode
mytarget_callback2(["micro_clean_existing"], [])
Waiting for bootloader device
Waiting for the new upload port...
PORT: COM5
"C:\Users\Remium\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages/tool-avrdude/avrdude" -C "C:\Users\Remium\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages/tool-avrdude/avrdude.conf" -p atmega32u4 -P COM5 -c avr109 -e
avrdude: AVR device initialized and ready to accept instructions
ignature = 0x1e9587 (probably m32u4)
avrdude: erasing chip
avrdude done.  Thank you.
Processing sparkfun_promicro_16 (board: sparkfun_promicro16; platform: atmelavr; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing felis/USB-Host-Shield-20 @ ^1.6.0
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: USB-Host-Shield-20@1.6.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/sparkfun_promicro16.html
PLATFORM: Atmel AVR (5.0.0) > SparkFun Pro Micro 5V/16MHz
HARDWARE: ATMEGA32U4 16MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) External (simavr)
PACKAGES: 
 - framework-arduino-avr @ 5.1.0 
 - tool-avrdude @ 1.70200.0 (7.2.0) 
 - toolchain-atmelavr @ 3.70300.220127 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- USB-Host-Shield-20 @ 1.6.0
|-- LUFA
|-- fxpt_math
Building in release mode
Compiling .pio\build\sparkfun_promicro_16\src\avr\lufa\LUFA.c.o
Compiling .pio\build\sparkfun_promicro_16\src\avr\lufa\reboot.c.o
Compiling .pio\build\sparkfun_promicro_16\src\avr\micro\main\main.cpp.o
Compiling .pio\build\sparkfun_promicro_16\src\avr\micro\shared\bootloader.c.o
Compiling .pio\build\sparkfun_promicro_16\src\avr\shared\io.cpp.o
Compiling .pio\build\sparkfun_promicro_16\src\avr\shared\pins.cpp.o
Compiling .pio\build\sparkfun_promicro_16\src\avr\shared\usb.cpp.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\config\config.c.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\controllers\controllers.c.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\hid\hid.cpp.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\main\adxl345.cpp.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\main\i2c_shared.cpp.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\main\max1704x.cpp.o
In file included from C:\Users\Remium\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages\framework-arduino-avr\libraries\LUFA/LUFA/LUFA/Drivers/USB/Core/ConfigDescriptors.h:56:0,
                 from C:\Users\Remium\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages\framework-arduino-avr\libraries\LUFA/LUFA/LUFA/Drivers/USB/USB.h:389,
                 from C:\Users\Remium\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages\framework-arduino-avr\libraries\LUFA/LUFA.c.inc:43,
                 from src\avr\lufa\LUFA.c:3:
C:\Users\Remium\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages\framework-arduino-avr\libraries\LUFA/LUFA/LUFA/Drivers/USB/Core/HostStandardReq.h:281:20: warning: 'USB_Host_SendControlRequest_PRV' declared 'static' but never defined [-Wunused-function]
     static uint8_t USB_Host_SendControlRequest_PRV(void* const BufferPtr);
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Remium\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages\framework-arduino-avr\libraries\LUFA/LUFA/LUFA/Drivers/USB/Core/HostStandardReq.h:282:20: warning: 'USB_Host_WaitForIOS' declared 'static' but never defined [-Wunused-function]
     static uint8_t USB_Host_WaitForIOS(const uint8_t WaitType);
                    ^~~~~~~~~~~~~~~~~~~
Compiling .pio\build\sparkfun_promicro_16\src\shared\main\mpr121.cpp.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\main\ps2.cpp.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\main\ps2_emu.cpp.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\main\shared_main.cpp.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\main\wii.cpp.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\main\wii_emu.cpp.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\usb\descriptors.cpp.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\usb\string_descriptors.c.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\usb\xbox.c.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\usb\xsm3\excrypt_des.c.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\usb\xsm3\excrypt_parve.c.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\usb\xsm3\excrypt_sha.c.o
Compiling .pio\build\sparkfun_promicro_16\src\shared\usb\xsm3\usbdsec.c.o
In file included from include/config.h:10:0,
                 from include/pin_funcs.h:2,
                 from include/shared_main.h:1,
                 from src\shared\main\shared_main.cpp:1:
src\shared\main\shared_main.cpp: In function 'uint8_t tick_inputs(void*, USB_LastReport_Data_t*, uint8_t)':
include/config_data.h:52:65: error: expected primary-expression before ')' token
     }if (((PINB & (1 << 6)) == 0  && (((!COMBINED_DEBOUNCE) || ())))) {\
                                                                 ^
src\shared\main\shared_main.cpp:1673:5: note: in expansion of macro 'TICK_SHARED'
     TICK_SHARED;
     ^~~~~~~~~~~
include/config_data.h:55:64: error: expected primary-expression before ')' token
     if (((PINB & (1 << 5)) == 0  && (((!COMBINED_DEBOUNCE) || ())))) {\
                                                                ^
src\shared\main\shared_main.cpp:1673:5: note: in expansion of macro 'TICK_SHARED'
     TICK_SHARED;
     ^~~~~~~~~~~
src\shared\main\shared_main.cpp:1794:10: warning: unused variable 'updateSequence' [-Wunused-variable]
     bool updateSequence = false;
          ^~~~~~~~~~~~~~
Compiling .pio\build\sparkfun_promicro_16\src\shared\usb\xsm3\xsm3.c.o
src\shared\main\shared_main.cpp: At global scope:
src\shared\main\shared_main.cpp:51:3: warning: 'rtt_t' defined but not used [-Wunused-variable]
 } rtt_t;
   ^~~~~
src\shared\main\shared_main.cpp:40:3: warning: 'ltt_t' defined but not used [-Wunused-variable]
 } ltt_t;
   ^~~~~
Compiling .pio\build\sparkfun_promicro_16\liba6d\SPI\SPI.cpp.o
*** [.pio\build\sparkfun_promicro_16\src\shared\main\shared_main.cpp.o] Error 1
========================== [FAILED] Took 2.33 seconds ==========================

### Version

latest and latest -1

### What operating system are you using?

Windows 11

### What microcontroller are you using?

arduino leonardo 5V

### How is your controller programmed?

Direct

sanjay900, 2024-08-14:
Thought I had fixed this one but I guess not. Not going to be able to properly fix it but for now you could probably get away with turning on combined strum debounce for the moment

Remium599, 2024-08-14:
Thanks, I managed to get around it by using a version from Last year. all is working fine at the moment

sanjay900, 2024-08-24:
Should be fixed now

closed 2024-08-24 by sanjay900