Santroller/Santroller #209: [Issue]: Bluetooth Receiver mode

Vitari-Translates, 2024-08-18:
### Please describe your issue.

In bluetooth receiver mode the software wont find any device.

### Version

v10.1.153

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

other issue not controller related or I have not gotten this far

UVClay, 2024-08-18:
Where did you get your Picos from?

Vitari-Translates, 2024-08-18:
> Where did you get your Picos from?

From a Hungarian Raspberry Shop.

sanjay900, 2024-08-19:
When you say the software won't find any device, what do you mean?
Like it's not showing up as something to program in santroller, or the receiver isn't detecting the transmitter

Vitari-Translates, 2024-08-19:
[image]
When i press Start Scan not found any device only the the "No Device Found". If I'm guessing correctly, this should work by listing nearby bluetooth devices.

sanjay900, 2024-08-19:
Specifically, only another pi pico w running the transmitter firmware

Vitari-Translates, 2024-08-19:
Sorry. Then I seem to have misunderstood its operation. Are there any plans to handle a bluetooth gamepad input?

sanjay900, 2024-08-20:
Potentially in the future, it'll probably be a while before I get around to that though.

closed 2024-08-20 by sanjay900

sanjay900, 2024-09-28:
I am looking into this at the moment. May i ask what kind of controller you were wanting to use?

Vitari-Translates, 2024-09-28:
Technically all of them. But I know it's not that simple. In the first
round probably the most popular ones like PS4,PS5, Nintendo Switch, XBOX
ONE.

Sanjay Govind ***@***.***> ezt írta (időpont: 2024. szept.
28., Szo, 23:37):

Vitari-Translates, 2024-09-28:
Technically all of them. But I know it's not that simple. In the first
round probably the most popular ones like PS4,PS5, Nintendo Switch, XBOX
ONE.

sanjay900, 2024-09-29:
certainly still needs work and i have zero clue what works and doesn't right now, but with the latest release a DS4 in pairing mode will connect as long as you remove bluetooth inputs, add it again and set the mode to classic.
Ill have to figure out how to make it stay connected permanently later.