Santroller/Santroller #211: [Issue]:

BionicPredator, 2024-08-26:
### Please describe your issue.

Installed the wii emulation file like it was said to do and when I went to start the game it loaded code and then froze. What went wrong or what did I do wrong? 

### Version

?

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-08-27:
honestly its all still a work in progress and doesn't really work with every game yet

sanjay900, 2024-12-18:
Oh by the way, this is all fixed now. The wii emulation stuff is fully stable.
https://github.com/Santroller/santroller-bslug/releases/latest

closed 2024-12-18 by sanjay900