Santroller/Santroller #212: [Issue]: Rock Band Pro Drums: Pads hitting multiple buttons

PilchDragon, 2024-08-27:
### Please describe your issue.

When i hit a pad on my drum set, a lot of outputs are triggered at the same time. Depending on the pad hit, these can trigger anywhere from 3 to 12 different outputs. These include various buttons, as well as stick and slider outputs. I am using YARG to test this, but it also does this on an Xbox 360.

I have followed the instructions on your website in order to wire up a drum set. I am currently using an Alesis Turbo Mesh kit.

I'll try to provide more details if required

### Version

10.1.165

### What operating system are you using?

Linux

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-08-28:
odd, not sure why that would be happening

sanjay900, 2024-08-28:
Save your current configration using the Save Settings button, and then send the following file:
Windows: `%appdata%/SantrollerConfigurator/Santroller/include/config_data.h`
Linux: `~/.config/SantrollerConfigurator/Santroller/include/config_data.h`
macOS: `/Users/Shared/Library/Application Support/SantrollerConfigurator/Santroller/include/config_data.h`

PilchDragon, 2024-08-28:
This is the config_data.h from the latest attempt at building the code [config_data.txt](https://github.com/user-attachments/files/16781192/config_data.txt)

sanjay900, 2024-08-28:
each drum hit will register both some analog value (via stick / slider) and a button. Once you have the minimums set correctly, that should stop a single drum hit registering as multiple pads, as right now hitting one pad is likley vibrating the kit enough that its registering tiny amounts on the other pads, and you have your minimum set to 0 right now, which makes it trigger on any hit.

PilchDragon, 2024-08-28:
I had not done this as I had assumed this would have been handled by the drum module. However, I have now tried setting a minimum threshold to 10,000 each. I still get the same result

PilchDragon, 2024-08-28:
In the Santroller GUI, only the correct pad is showing as being hit, I'm not seeing any others registering a hit

sanjay900, 2024-08-28:
when you say 3 - 12 buttons, what buttons specifically are being hit at the same time?

PilchDragon, 2024-08-28:
Using YARG, it reports the following buttons when the red pad is struck:

- Button 2
- Button 15
- Rx
- Ry
- Z
- Rz
- Slider
- Stick X
- Stick Y
- Stick Down
- Stick Left
- Stick Right

Weirdly, I tried again immediately after and it reported these:
- Button 2
- Button 15
- Stick Y

The other pads act in a similar manner, but the first button is different each time (eg. Yellow pad reports button 5 instead of button 2)

sanjay900, 2024-08-28:
okay, the axis doing that seems odd but it is expected that multiple actual buttons are reported for a hit
im wondering if that first time YARG was just receiving an initial report where all the axis` were being read for the first time or something.

PilchDragon, 2024-08-28:
Would that be a clue as to why it seems to be playing up on 360 RB3DX too? I'll set up the 360 and see if that provides any additional information

sanjay900, 2024-08-28:
have you tried RB3DX after you changed the minimum away from 0?

PilchDragon, 2024-08-28:
Not yet, I'll try that now

sanjay900, 2024-08-28:
[firmware.uf2.zip](https://github.com/user-attachments/files/16781914/firmware.uf2.zip)
can also see if this makes any difference

PilchDragon, 2024-08-28:
Its possible that it might be working now. I'll do some more testing just in case (yellow cymbal is reporting as blue for some reason)

sanjay900, 2024-08-28:
might just mean that the default mappings don't quite align with your drum kit, but you can always disable the defaults and assign the channels yourself to flip them around

sanjay900, 2024-08-28:
comparing to my notes i have them the right way around, so ill just assume that my midi note values dont align with your kit

PilchDragon, 2024-08-28:
Just tried it on the 360, it looks like its all working correctly now. I have a feeling it was the sensitivities all being set to 0 that was throwing it off. Thank you for all of your help with resolving this

closed 2024-08-28 by PilchDragon

sanjay900, 2024-08-28:
I've pushed a new release that defaults the sensitivties to a non zero value to hopefully help with this in the future for the next person

PilchDragon, 2024-08-28:
Sounds like a good idea. Thanks again for your help :)