Santroller/Santroller #217: [Issue]: Harmonix Drum Controller for Nintendo Wii (USB Host)

sfurtaw, 2024-08-31:
### Please describe your issue.

In USB Host mode, Harmonix Drum Controller for Nintendo Wii reads as PS3 Rock Band Drums, bindings do not work
Device ID: HID\VID_1BAD&PID_0005&REV_1000

The Pi Pico is confirmed working in USB host mode with Wii guitars (via adapter) and PS4 controllers.

### Version

10.1.175

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

other issue not controller related or I have not gotten this far

sanjay900, 2024-08-31:
Did you add the caps? This is the specific scenario they are necessary for

sfurtaw, 2024-09-01:
I do have two 47pF caps between D+ and a ground pin, and D- and a ground pin... I will have access to my multimeter tomorrow to double check.

I know my drum kit does not have a solder pad that converts it into a PS3 kit...... although some of the Wii ones do. It's interesting that they're still the same.

sanjay900, 2024-09-01:
how odd then. Do you see an input if you hold the button down as you plug it into the pico?

sfurtaw, 2024-09-02:
It definitely seems to be doing the input thing when I plug it in - I did some testing it it does seem like a cap problem, yeah.

Thanks for the help 👍👍

sanjay900, 2024-09-21:
It also may just work without the caps and resistors on the latest release as well

closed 2024-09-21 by sanjay900