Santroller/Santroller #218: [Request]: Add a threshold setting for MIDI Keys Modulation Wheel

LinosM, 2024-09-02:
### Please describe your request.

My Alesis Keytar for example has a tilt function that controls modulation, it'd be cool to set a threshold so it can deploy overdrive at a certain angle instead of at any point.

[image]

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

N/A

### How will the microcontroller be programed for this request?

N/A

sanjay900, 2024-09-16:
How does RB decide what dictates overdrive?

LinosM, 2024-09-16:
I'm not sure I understand the question.

The Midi Pro Adapter doesn't utilize the modulation wheel at all so there's no way to dictate how it deploys, and the current implementation of the mod wheel in Santroller would just have it deploy whenever the value changes to any number. I'm asking for an optional setting so it doesn't trigger until the value reaches above a user-specified value.

sanjay900, 2024-09-17:
Oh right yeah, i can add a threshold for sure

sanjay900, 2024-09-20:
Should be resolved with v10.1.202

closed 2024-09-20 by sanjay900