Santroller/Santroller #219: [Request]: PS2_emu files

sonik-br, 2024-09-03:
### Please describe your request.

Hi,
Not exactly a feature request...

I'm trying to build a firmware with just the ps2_emu code from Santroller. My idea is to expand it to emulate some alternative ps1 controllers (ie: guncon) using a pico.

But I could not find where some macros are defined, like this one:

https://github.com/Santroller/Santroller/blob/master/src/shared/main/shared_main.cpp#L1555

Also, interesting idea to disable one of the SPI pins here:
https://github.com/Santroller/Santroller/blob/master/src/shared/main/ps2_emu.cpp#L20

But I could not understand exactly how it works.
By disabling the MOSI pin, the pico will not receive data but will output data. Is that right?

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

PS2

sanjay900, 2024-09-03:
The PS2 firmware is a bit borked at the moment, i want to rewrite it.
None of the PS2 hacks will make any sense because they are work arounds because the picos SPI hardware is incompatible with how the PS2 works.
https://github.com/Loc15/PicoGamepadConverter
I would recommend looking at a project like this, because these just use PIO to emulate the PS2 signals, which is what i will be moving santroller to eventually when i get the time.

sanjay900, 2024-09-03:
All those definitions you can't find come from the santroller tool itself, build a firmware using santroller and then you will be able to find what you need that way, though santroller as a firmware really was not designed to be decoupled from the tooling.

closed 2024-09-03 by sanjay900

sonik-br, 2024-09-04:
Ok, thanks.

I've been using PIO for this but all current solutions (based on picomemcard) are not reliable.
They tend to hang or stop working when changing games on a ps2 via OPL.
I still need to debug this and understand why it hangs.

sanjay900, 2024-09-04:
Interesting, does my solution somehow not do that?

sonik-br, 2024-09-04:
I'm not sure, still have to test.
Right now I have a ps2 connector to output, but no way to do any input to pico. Need to grab a usb-a female connector to use as usb host.