Santroller/Santroller #221: [Request]: wii Rock band keytar to x-input (for 360)

kimo1arquimedes1, 2024-09-19:
### Please describe your request.

Hello, I recently acquired an RB Wii keytar, but I want to use it on an Xbox 360. However, after reviewing the list of compatible USB devices, this controller doesn't appear. I tried connecting the dongle to the Raspberry Pi Pico, but I wasn't successful. It detects it as a generic controller, and I tried all the generic controller options without any luck. I'd like to know if there’s a way to add support to convert the Wii keytar to X-input and use it on the 360.

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

N/A

sanjay900, 2024-09-19:
Can you run this and get me the details of the adapter
https://www.thesycon.de/eng/usb_descriptordumper.shtml

kimo1arquimedes1, 2024-09-19:
Information for device Harmonix RB3 Keyboard for Nintendo Wii (VID=0x1BAD PID=0x3330):

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device address: 0x0004
Current configuration value: 0x01
Number of open pipes: 2

------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0110	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x08	bMaxPacketSize0   (8 bytes)
0x1BAD	idVendor
0x3330	idProduct
0x0005	bcdDevice
0x01	iManufacturer   "Licensed by Nintendo of America "
0x02	iProduct        "Harmonix RB3 Keyboard for Nintendo Wii"
0x00	iSerialNumber
0x01	bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0029	wTotalLength   (41 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0x80	bmAttributes   (Bus-powered Device)
0x32	bMaxPower      (100 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0101	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x0089	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x0A	bInterval         (10 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x01	bInterval         (1 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "Licensed by Nintendo of America "
0x02   0x0409  "Harmonix RB3 Keyboard for Nintendo Wii"

------------------------------

Connection path for device: 
Controladora de host USB genérico compatible con xHCI
Root Hub
Concentrador USB genérico
Harmonix RB3 Keyboard for Nintendo Wii (VID=0x1BAD PID=0x3330) Port: 1

Running on: Windows 10 or greater (Build Version 19045)

Brought to you by TDD v2.19.0, Dec  5 2023, 12:08:38

_______________Thanks for replying, here's the info

sanjay900, 2024-09-20:
Yeah okay i just don't have it implemented right now ill do that when i get a chance.

kimo1arquimedes1, 2024-09-20:
Thank you, I'll be waiting for news

sanjay900, 2024-09-20:
Hopefully it works with v10.1.203
You will need to remove and then re-add usb host inputs for it to pick up the new inputs

kimo1arquimedes1, 2024-09-27:
Sorry, thank you for following up on my request. I was away for a week due to academic matters and couldn't thank you earlier.

Currently, I’ve tried several versions, from v10.1.204 to v10.1.227, but I haven’t been successful in using my Rock Band keytar.

1: The controller is detected as a PS3 gamepad through the USB host.
2: When I try to save changes, I get an error using Xinput and Keyboard Pro. I’ve tried with several Raspberry Pi Pico boards, but the error remains the same.
3: When I try to check the available buttons on the controller, it doesn’t detect them. Some buttons, like X and Start, appear as held down, but no other keys are detected. I’ve tried all modifications.

Thank you. Kind regards from Mexico

sanjay900, 2024-09-27:
What error?

sanjay900, 2024-09-27:
Save your current configuration using the `Save Settings` button, and then send the following file:
Windows: `%appdata%/SantrollerConfigurator/Santroller/include/config_data.h`
Linux: `~/.config/SantrollerConfigurator/Santroller/include/config_data.h`
macOS: `/Users/Shared/Library/Application Support/SantrollerConfigurator/Santroller/include/config_data.h`

XxyZwW, 2024-09-27:
Hello, I am in the same situation and interest, I would like to contribute in whatever way I can to help, so I also tried with the latest version and it also throws me the problem, I tried multiple configurations but the most logical ones that I provided and relevant to mention are the following:
Test 1
1.- Program the microcontroller as: input type - USB Adapter
2.- In the configuration, controller settings section - controller type - Pro Keys
3.- In USB Host Inputs - I assign the pins of the USB connected to the Pi pico
Result: The following error appears

Test 2
1.- Program the microcontroller as: input type - Wii Adapter
2.- In the configuration, controller settings section - controller type - Pro Keys
3.- Assign the SDA pin and SCL pin and save settings
4.- Add settings - USB Host Inputs
3.- In USB Host Inputs - I assign the pins of the USB connected to the Pi pico
Result: Arrjoa the following error

[image]
[image]
[image]
[image]

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/9e55f6db5c56b9867c69fe473f388beea4546672.zip@1.14.0; board: pico; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.14.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, pico-debug, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.40002.0 (4.0.2) 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-picotool-rp2040-earlephilhower @ 5.120300.240827 (12.3.0) 
 - toolchain-rp2040-earlephilhower @ 5.120300.240125 (12.3.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 67 compatible libraries
Scanning dependencies...
Dependency Graph
|-- USB-Host-Shield-20 @ 1.6.2
|-- MIDI Library @ 5.0.2
|-- SPI @ 1.0
|-- Wire @ 1.0
|-- fxpt_math
Building in release mode
Compiling .pio\build\pico\src\pico\bt.cpp.o
Compiling .pio\build\pico\src\pico\bt_rx.cpp.o
Compiling .pio\build\pico\src\pico\io.cpp.o
Compiling .pio\build\pico\src\pico\main.cpp.o
Compiling .pio\build\pico\src\pico\pins.cpp.o
Compiling .pio\build\pico\src\pico\xinput_device.cpp.o
Compiling .pio\build\pico\src\pico\xinput_host.cpp.o
Compiling .pio\build\pico\src\shared\config\config.c.o
Compiling .pio\build\pico\src\shared\controllers\controllers.c.o
Compiling .pio\build\pico\src\shared\hid\hid.cpp.o
Compiling .pio\build\pico\src\shared\main\accel.cpp.o
Compiling .pio\build\pico\src\shared\main\max1704x.cpp.o
Compiling .pio\build\pico\src\shared\main\mpr121.cpp.o
Compiling .pio\build\pico\src\shared\main\ps2.cpp.o
Compiling .pio\build\pico\src\shared\main\ps2_emu.cpp.o
Compiling .pio\build\pico\src\shared\main\shared_main.cpp.o
Compiling .pio\build\pico\src\shared\main\wii.cpp.o
Compiling .pio\build\pico\src\shared\main\wii_emu.cpp.o
Compiling .pio\build\pico\src\shared\usb\descriptors.cpp.o
Compiling .pio\build\pico\src\shared\usb\string_descriptors.c.o
Compiling .pio\build\pico\src\shared\usb\xbox.c.o
In file included from include/config.h:10,
[... 1665 lines cut ...]
      |             ^~~~~~~~
include/config_data.h:387:46: error: 'struct USB_Host_Data_t' has no member named 'proKey'
  387 |         report->pedalDigital = usb_host_data.proKey > 1;\
      |                                              ^~~~~~
src\shared\main\shared_main.cpp:2085:13: note: in expansion of macro 'TICK_PS3'
 2085 |             TICK_PS3;
      |             ^~~~~~~~
include/config_data.h:389:23: error: 'struct USB_Host_Data_t' has no member named 'proKey'
  389 |     if (usb_host_data.proKey) {\
      |                       ^~~~~~
src\shared\main\shared_main.cpp:2085:13: note: in expansion of macro 'TICK_PS3'
 2085 |             TICK_PS3;
      |             ^~~~~~~~
include/config_data.h:390:96: error: 'struct USB_Host_Data_t' has no member named 'proKey'
  390 |         report->touchPad = handle_calibration_ps3_360_trigger(report->touchPad, (usb_host_data.proKey), 0, 512, 0);\
      |                                                                                                ^~~~~~
src\shared\main\shared_main.cpp:2085:13: note: in expansion of macro 'TICK_PS3'
 2085 |             TICK_PS3;
      |             ^~~~~~~~
*** [.pio\build\pico\src\shared\main\shared_main.cpp.o] Error 1
========================== [FAILED] Took 6.73 seconds ==========================

The log is from some previous attempts.

By the way, when the error occurs, the whole Windows slows down, the explorer and the taskbar constantly restart.

XxyZwW, 2024-09-27:
I also attach my details of the Harmonix Keyboard adapter obtained with the tool you mentioned usb_descriptordumper, in case it helps or contributes something

Thank you very much for the work you do

Information for device Harmonix RB3 Keyboard for Nintendo Wii (VID=0x1BAD PID=0x3330):

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device address: 0x001A
Current configuration value: 0x01
Number of open pipes: 2

------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0110	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x08	bMaxPacketSize0   (8 bytes)
0x1BAD	idVendor
0x3330	idProduct
0x0005	bcdDevice
0x01	iManufacturer   "Licensed by Nintendo of America "
0x02	iProduct        "Harmonix RB3 Keyboard for Nintendo Wii"
0x00	iSerialNumber
0x01	bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0029	wTotalLength   (41 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0x80	bmAttributes   (Bus-powered Device)
0x32	bMaxPower      (100 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0101	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x0089	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x0A	bInterval         (10 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x01	bInterval         (1 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "Licensed by Nintendo of America "
0x02   0x0409  "Harmonix RB3 Keyboard for Nintendo Wii"

------------------------------

Connection path for device: 
Controladora de host USB genérico compatible con xHCI
Root Hub
Concentrador USB genérico
Concentrador USB genérico
Concentrador USB genérico
Harmonix RB3 Keyboard for Nintendo Wii (VID=0x1BAD PID=0x3330) Port: 1

Running on: Windows 10 or greater (Build Version 22635)

Brought to you by TDD v2.19.0, Dec  5 2023, 12:08:38

sanjay900, 2024-09-27:
v10.1.228 should pick the keyboards up, and should fix that error.

sanjay900, 2024-09-27:
It should be out now. Let me know if the mappings are actually working or not.

kimo1arquimedes1, 2024-09-28:
Thanks, it works wonderfully, but there are two issues.

---The Overdrive button wasn’t detected by Rock Band 3, so I had to add it manually, which resolved the problem. (It's worth mentioning that when adding the Overdrive button, the option to adjust the limit appeared, and I set it to 0).
---The touch effect panel, even after adding it manually, is still not detected by the game (I should note that in this case, the option to adjust the limit does not appear).

Both buttons are detected and even light up when activated in the app. However, the game doesn’t recognize them, and one was adjusted in the way I explained earlier.

sanjay900, 2024-09-28:
can you send me the config_data.h file without your manual overdrive

sanjay900, 2024-09-28:
Try v10.1.230 when it comes out. However, you will need to remove usb host inputs and then add it again.

kimo1arquimedes1, 2024-09-28:
Perfect, thanks for the attention. 100% functional keytar

closed 2024-09-28 by sanjay900