Santroller/Santroller #223: [Issue]: Slowdowns and conflicts with Windows when running Santroller

XxyZwW, 2024-09-27:
### Please describe your issue.

Hello, I have noticed that from version 10.1.153 onwards, when the tool displays a window with instructions or error messages in the microcontroller programming, it causes Windows to slow down and even the explorer and the start bar to close and open constantly, until I close Santroller the OS stabilizes again, perhaps this is a problem with my OS, but I would like to confirm since previously this did not happen to me when running Santroller.
I leave a link to a video of how the problem occurs.
[Video](https://drive.google.com/file/d/1Aw7meIfxHgG5M8-4vl-mY2pd9S7o_uQN/view?usp=sharing)

### Version

10.1.153 onwards

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

other issue not controller related or I have not gotten this far

sanjay900, 2024-09-27:
Im not really sure why specifically 153 would be where the problems start, i was expecting maybe 191 but i dont think i changed anything with 153 on that front

sanjay900, 2024-09-27:
Does it happen with any of the versions before 153?

sanjay900, 2024-09-27:
I unfortunately have to assume this problem is specific to your computer, as I haven't seen anybody else report it. Might be worth deleting everything inside of `%appdata%/SantrollerConfigurator/Santroller` besides `config.bin` and seeing if that makes a difference.

XxyZwW, 2024-09-29:
You're right, I did some more tests and it seems to be my OS, I haven't been able to solve it but from the results it seems that Santroller is not the problem, thank you very much.

closed 2024-09-29 by XxyZwW