Santroller/Santroller #224: [Issue]: 'bt_data' was not declared in this scope when building for Bluetooth Receiver on Pi Pico

o0kryos0o, 2024-09-29:
### Please describe your issue.

```c
-Host-Shield-20\Wii.cpp.o
Compiling .pio\build\picow\lib324\USB-Host-Shield-20\XBOXOLD.cpp.o
src\shared\hid\hid.cpp: In function 'uint8_t handle_serial_command(uint8_t, uint16_t, uint8_t*, bool*)':
src\shared\hid\hid.cpp:1059:38: error: 'bt_data' was not declared in this scope
 1059 |             memcpy(response_buffer, &bt_data, sizeof(bt_data));
      |                                      ^~~~~~~
*** [.pio\build\picow\src\shared\hid\hid.cpp.o] Error 1
Compiling .pio\build\picow\lib324\USB-Host-Shield-20\XBOXONE.cpp.o
src\pico\bt_rx.cpp: In function 'void handle_gatt_client_event(uint8_t, uint16_t, uint8_t*, uint16_t)':
src\pico\bt_rx.cpp:316:96: error: 'bt_data' was not declared in this scope
  316 |                 fill_generic_report(info, gattservice_subevent_hid_report_get_report(packet), &bt_data);
      |                                                                                                ^~~~~~~
*** [.pio\build\picow\src\pico\bt_rx.cpp.o] Error 1
```

### Version

10.1.240

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-09-30:
Try deleting everying in `%appdata%/SantrollerConfigurator` besides config.bin

sanjay900, 2024-10-04:
fixed with v10.1.241

closed 2024-10-04 by sanjay900