Santroller/Santroller #225: [Issue]: rashberry pi pico, Error when starting programming in any mode with the latest update (USB host, direct wiring)

kimo1arquimedes1, 2024-09-29:
### Please describe your issue.

I've been trying to program 2 rashberry pi pico and both throw this error with the latest updates. this did not occur in the SantrollerConfigurator v10.1.232 version

[image]

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/9e55f6db5c56b9867c69fe473f388beea4546672.zip@1.14.0; board: pico; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing https://github.com/Santroller/USB_Host_Shield_2.0/archive/refs/heads/master.zip @ 1.6.2
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: USB-Host-Shield-20@1.6.2 has been installed!
Library Manager: Installing fortyseveneffects/MIDI Library @ 5.0.2
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: MIDI Library@5.0.2 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.14.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, pico-debug, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.40002.0 (4.0.2) 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-picotool-rp2040-earlephilhower @ 5.120300.240827 (12.3.0) 
 - toolchain-rp2040-earlephilhower @ 5.120300.240125 (12.3.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 67 compatible libraries
Scanning dependencies...
Dependency Graph
|-- USB-Host-Shield-20 @ 1.6.2
|-- MIDI Library @ 5.0.2
|-- SPI @ 1.0
|-- Wire @ 1.0
|-- fxpt_math
Building in release mode
Compiling .pio\build\pico\FrameworkArduinoBootloader\boot2_w25q080_2_padded_checksum.S.o
Compiling .pio\build\pico\src\pico\bt.cpp.o
Compiling .pio\build\pico\src\pico\bt_rx.cpp.o
Compiling .pio\build\pico\src\pico\bt_rx_cnt.cpp.o
Compiling .pio\build\pico\src\pico\hidparser.c.o
Compiling .pio\build\pico\src\pico\io.cpp.o
Compiling .pio\build\pico\src\pico\main.cpp.o
Compiling .pio\build\pico\src\pico\midi_host.c.o
Compiling .pio\build\pico\src\pico\pins.cpp.o
In file included from c:\users\kishi\onedrive\documentos\santrollerconfigurator-win-portable\current\binaries\platformio\packages\toolchain-rp2040-earlephilhower\arm-none-eabi\include\c++\12.3.0\stdlib.h:36,
                 from C:\Users\kishi\OneDrive\Documentos\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico\cores\rp2040/Arduino.h:24,
                 from src\pico\main.cpp:1:
c:\users\kishi\onedrive\documentos\santrollerconfigurator-win-portable\current\binaries\platformio\packages\toolchain-rp2040-earlephilhower\arm-none-eabi\include\c++\12.3.0\cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory
   41 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from C:\Users\kishi\OneDrive\Documentos\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base_headers/include/pico/assert.h:14,
                 from C:\Users\kishi\OneDrive\Documentos\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base_headers/include/pico/types.h:12,
                 from C:\Users\kishi\OneDrive\Documentos\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base_headers/include/pico.h:26,
                 from C:\Users\kishi\OneDrive\Documentos\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/rp2_common/hardware_adc/include/hardware/adc.h:10,
                 from src\pico\pins.cpp:1:
c:\users\kishi\onedrive\documentos\santrollerconfigurator-win-portable\current\binaries\platformio\packages\toolchain-rp2040-earlephilhower\arm-none-eabi\include\c++\12.3.0\cassert:43:10: fatal error: bits/c++config.h: No such file or directory
   43 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\pico\src\pico\pins.cpp.o] Error 1
*** [.pio\build\pico\src\pico\main.cpp.o] Error 1
In file included from C:\Users\kishi\OneDrive\Documentos\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base_headers/include/pico/assert.h:14,
                 from C:\Users\kishi\OneDrive\Documentos\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base_headers/include/pico/types.h:12,
                 from C:\Users\kishi\OneDrive\Documentos\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/common/pico_base_headers/include/pico.h:26,
                 from C:\Users\kishi\OneDrive\Documentos\SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico/pico-sdk/src/rp2_common/hardware_spi/include/hardware/spi.h:10,
                 from src\pico/io_define.h:2,
                 from include/io.h:4,
                 from src\pico\io.cpp:1:
c:\users\kishi\onedrive\documentos\santrollerconfigurator-win-portable\current\binaries\platformio\packages\toolchain-rp2040-earlephilhower\arm-none-eabi\include\c++\12.3.0\cassert:43:10: fatal error: bits/c++config.h: No such file or directory
   43 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\pico\src\pico\io.cpp.o] Error 1
========================= [FAILED] Took 12.88 seconds =========================

### Version

SantrollerConfigurator v10.1.240

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

kimo1arquimedes1, 2024-09-29:
[image]

sanjay900, 2024-10-01:
Can you try running it from somewhere else?
Usually this error means it doesn't like being run from the location you have it installed

closed 2024-10-02 by kimo1arquimedes1