Santroller/Santroller #227: [Request]: RP2040 Zero Support for WS2812 RGB LEDs on GPIO 16

XxyZwW, 2024-10-04:
### Please describe your request.

I have a waveshare RP2040 Zero, it's basically a stripped-down Pi pico, it has almost all the same GPIOs as the Pi pico, in fact, I've tested it with some configurations with Santroller in different modes and it works very well, only the integrated LED on the board is assigned to GPIO 16, I tried to control this LED from Santroller, but I haven't been successful, and in the LED Settings section when choosing WS2812 RGB the GPIO16 doesn't appear in the Data Pin list, would it be possible to add support for this board and GPIO 16?

Thank you very much for the work you do

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

N/A

### How will the microcontroller be programed for this request?

N/A

sanjay900, 2024-10-04:
Unfortunately there are reasons why that's the case, it would need a rewrite to support any other pins

sanjay900, 2025-01-29:
This one is supported in v10.2.8

closed 2025-01-29 by sanjay900