Santroller/Santroller #229: [Request]: Support for Fender Mustang Pro Guitar Controller

XxyZwW, 2024-10-16:
### Please describe your request.

Is it possible to add support for the Rock Band Fender Mustang Pro Guitar Controller for Wii to use on Xbox 360? This has a lot of little buttons all around the neck and strings on the body of the guitar and this comes with a USB dongle like the one on the Rock Band keyboard

I haven't been able to try it yet and I'm really curious to use it, I think this enables pro mode in the Rock Band 3 game.

I see that in the Santroller tool there is no PRO guitar section in any of the menus other than the Fornite Festival section, but I think the gameplay is different.

[image]

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Wii

sanjay900, 2024-10-23:
may work now, haven't got one to test

XxyZwW, 2024-10-24:
Hello, I tried to program with the latest version, but the tool gets stuck on writing the device.
[image]
[image]
Since the "Refresh Device" button does not appear in the main window of the tool, I tried to open an additional window as it was done before to be able to view and use the button, but now I receive the message that there is already a Santroller instance running and it does not open the second window where I could use the button.
[image]
I also tried to program the device first as a Gamepad to later enter the configuration and change it to the Mustang Pro Guitar, again it got stuck on writing, but once inside the configuration screen the "Refresh Device" button does appear. When I pressed it, it took a while to launch the window to disconnect and reconnect the device and I did so, but it continued to get stuck.
[image]

This is what I was able to try.
I thank you in advance for the attention you give us and the project, you do an extraordinary job.

XxyZwW, 2024-10-26:
I saw that for the keytar you asked the user to run usb_descriptordumper, would it help if I shared the reading I got when connecting the adapter to the PC?

Information for device Harmonix RB3 Mustang Guitar for Nintendo Wii (VID=0x1BAD PID=0x3430):

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device address: 0x0028
Current configuration value: 0x01
Number of open pipes: 2

------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0110	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x08	bMaxPacketSize0   (8 bytes)
0x1BAD	idVendor
0x3430	idProduct
0x0000	bcdDevice
0x01	iManufacturer   "Licensed by Nintendo of America "
0x02	iProduct        "Harmonix RB3 Mustang Guitar for Nintendo Wii"
0x00	iSerialNumber
0x01	bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0029	wTotalLength   (41 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0x80	bmAttributes   (Bus-powered Device)
0x32	bMaxPower      (100 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0101	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x0089	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x0A	bInterval         (10 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x01	bInterval         (1 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "Licensed by Nintendo of America "
0x02   0x0409  "Harmonix RB3 Mustang Guitar for Nintendo Wii"

------------------------------

Connection path for device: 
Controladora de host USB genérico compatible con xHCI
Root Hub
Concentrador USB genérico
Concentrador USB genérico
Harmonix RB3 Mustang Guitar for Nintendo Wii (VID=0x1BAD PID=0x3430) Port: 1

Running on: Windows 10 or greater (Build Version 22635)

Brought to you by TDD v2.19.0, Dec  5 2023, 12:08:38

XxyZwW, 2024-10-27:
Indeed, it no longer gets stuck in the process, I was able to program a pi pico device, but when trying to test it, it does detect the pro guitar on Xbox360, but none of the buttons work to move through the menus except for the right pad, which I think is the one assigned to the select button.
I would also like to mention that from Santroler when I configure the USB adapter for the Mustang Pro Guitar, in the "Pro Guitar Inputs" confirmation section no input appears.

[image]

Also, when I try to program with the "Use Passthrough with RPCS3" option disabled, Santroler does not complete the programming, throwing the following error:

> Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/9e55f6db5c56b9867c69fe473f388beea4546672.zip@1.14.0; board: pico; framework: arduino)
> --------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
> PLATFORM: Raspberry Pi RP2040 (1.14.0) > Raspberry Pi Pico
> HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
> DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, pico-debug, picoprobe, raspberrypi-swd)
> PACKAGES: 
>  - framework-arduinopico @ 1.40002.0 (4.0.2) 
>  - tool-openocd-rp2040-earlephilhower @ 0.0.0 
>  - tool-openocd-rp2040-earlephilhower @ 0.0.0 
>  - tool-picotool-rp2040-earlephilhower @ 5.120300.240827 (12.3.0) 
>  - toolchain-rp2040-earlephilhower @ 5.120300.240125 (12.3.0)
> Flash size: 2.00MB
> Sketch size: 2.00MB
> Filesystem size: 0.00MB
> Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Found 67 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- USB-Host-Shield-20 @ 1.6.2
> |-- MIDI Library @ 5.0.2
> |-- SPI @ 1.0
> |-- Wire @ 1.0
> |-- fxpt_math
> Building in release mode
> Compiling .pio\build\pico\src\pico\bt.cpp.o
> Compiling .pio\build\pico\src\pico\bt_rx.cpp.o
> Compiling .pio\build\pico\src\pico\bt_rx_cnt.cpp.o
> Compiling .pio\build\pico\src\pico\hidparser.c.o
> Compiling .pio\build\pico\src\pico\io.cpp.o
> Compiling .pio\build\pico\src\pico\main.cpp.o
> Compiling .pio\build\pico\src\pico\pins.cpp.o
> Compiling .pio\build\pico\src\pico\xinput_device.cpp.o
> Compiling .pio\build\pico\src\pico\xinput_host.cpp.o
> Compiling .pio\build\pico\src\shared\config\config.c.o
> Compiling .pio\build\pico\src\shared\controllers\controllers.c.o
> Compiling .pio\build\pico\src\shared\hid\hid.cpp.o
> Compiling .pio\build\pico\src\shared\main\accel.cpp.o
> Compiling .pio\build\pico\src\shared\main\max1704x.cpp.o
> Compiling .pio\build\pico\src\shared\main\mpr121.cpp.o
> Compiling .pio\build\pico\src\shared\main\ps2.cpp.o
> Compiling .pio\build\pico\src\shared\main\ps2_emu.cpp.o
> Compiling .pio\build\pico\src\shared\main\shared_main.cpp.o
> Compiling .pio\build\pico\src\shared\main\wii.cpp.o
> Compiling .pio\build\pico\src\shared\main\wii_emu.cpp.o
> Compiling .pio\build\pico\src\shared\usb\descriptors.cpp.o
> Compiling .pio\build\pico\src\shared\usb\string_descriptors.c.o
> Compiling .pio\build\pico\src\shared\usb\xbox.c.o
> Linking .pio\build\pico\firmware.elf
> Generating UF2 image
> picotool uf2 convert -t elf ".pio\build\pico\firmware.elf" ".pio\build\pico\firmware.uf2"
> Retrieving maximum program size .pio\build\pico\firmware.elf
> Flash size: 2.00MB
> Sketch size: 2.00MB
> Filesystem size: 0.00MB
> Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
> Checking size .pio\build\pico\firmware.elf
> Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
> RAM:   [==        ]  21.3% (used 55956 bytes from 262144 bytes)
> Flash: [=         ]   5.6% (used 116244 bytes from 2093056 bytes)
> Building .pio\build\pico\firmware.bin
> Configuring upload protocol...
> AVAILABLE: blackmagic, cmsis-dap, jlink, mbed, pico-debug, picoprobe, picotool, raspberrypi-swd
> CURRENT: upload_protocol = mbed
> Trying to reset Pico into bootloader mode...
> Looking for upload disk...
> Using manually specified: D:\
> Uploading .pio\build\pico\firmware.bin
> *** [upload] D:\firmware.uf2: Invalid argument
> ========================= [FAILED] Took 12.96 seconds =========================

Anything I can contribute or help with, I'm available.

sanjay900, 2024-12-24:
Does this work any better with the latest firmwares?

XxyZwW, 2024-12-24:
No, it keeps giving me an error when I try to program it
I just tried with the latest 10.1.308

[image]

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/39b90392af50585e429941bd2561a91949a2ba46.zip@1.15.0; board: pico; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing https://github.com/Santroller/USB_Host_Shield_2.0/archive/refs/heads/master.zip @ 1.6.3
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: USB-Host-Shield-20@1.6.3 has been installed!
Library Manager: Installing fortyseveneffects/MIDI Library @ 5.0.2
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: MIDI Library@5.0.2 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.15.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, pico-debug, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.40400.0 (4.4.0) 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-picotool-rp2040-earlephilhower @ 5.140200.240929 (14.2.0) 
 - toolchain-rp2040-earlephilhower @ 5.140200.240929 (14.2.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
PSRAM size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 70 compatible libraries
Scanning dependencies...
Dependency Graph
|-- USB-Host-Shield-20 @ 1.6.3
|-- MIDI Library @ 5.0.2
|-- SPI @ 1.0
|-- Wire @ 1.0
|-- fxpt_math
Building in release mode
Compiling .pio\build\pico\FrameworkArduinoBootloader\boot2_w25q080_2_padded_checksum.S.o
Compiling .pio\build\pico\src\pico\bt.cpp.o
Compiling .pio\build\pico\src\pico\bt_rx.cpp.o
Compiling .pio\build\pico\src\pico\bt_rx_cnt.cpp.o
Compiling .pio\build\pico\src\pico\hidparser.c.o
Compiling .pio\build\pico\src\pico\io.cpp.o
Compiling .pio\build\pico\src\pico\main.cpp.o
Compiling .pio\build\pico\src\pico\midi_host.c.o
Compiling .pio\build\pico\src\pico\pins.cpp.o
Compiling .pio\build\pico\src\pico\xinput_device.cpp.o
Compiling .pio\build\pico\src\pico\xinput_host.cpp.o
Compiling .pio\build\pico\src\shared\config\config.c.o
Compiling .pio\build\pico\src\shared\controllers\controllers.c.o
Compiling .pio\build\pico\src\shared\hid\hid.cpp.o
Compiling .pio\build\pico\src\shared\main\accel.cpp.o
Compiling .pio\build\pico\src\shared\main\i2c_shared.cpp.o
Compiling .pio\build\pico\src\shared\main\max1704x.cpp.o
Compiling .pio\build\pico\src\shared\main\mpr121.cpp.o
Compiling .pio\build\pico\src\shared\main\ps2.cpp.o
Compiling .pio\build\pico\src\shared\main\ps2_emu.cpp.o
Compiling .pio\build\pico\src\shared\main\shared_main.cpp.o
Compiling .pio\build\pico\src\shared\main\wii.cpp.o
Compiling .pio\build\pico\src\shared\main\wii_emu.cpp.o
Compiling .pio\build\pico\src\shared\usb\descriptors.cpp.o
Compiling .pio\build\pico\src\shared\usb\string_descriptors.c.o
Compiling .pio\build\pico\src\shared\usb\xbox.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\excrypt_des.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\excrypt_parve.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\excrypt_sha.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\usbdsec.c.o
[... 93 lines cut ...]
Linking .pio\build\pico\firmware.elf
Generating UF2 image
picotool uf2 convert -t elf ".pio\build\pico\firmware.elf" ".pio\build\pico\firmware.uf2"
Retrieving maximum program size .pio\build\pico\firmware.elf
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
PSRAM size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
Checking size .pio\build\pico\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  21.4% (used 56072 bytes from 262144 bytes)
Flash: [=         ]   5.9% (used 122896 bytes from 2093056 bytes)
Building .pio\build\pico\firmware.bin
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed, pico-debug, picoprobe, picotool, raspberrypi-swd
CURRENT: upload_protocol = mbed
Trying to reset Pico into bootloader mode...
Looking for upload disk...
Using manually specified: G:\
Uploading .pio\build\pico\firmware.bin
*** [upload] G:\firmware.uf2: Invalid argument
========================= [FAILED] Took 66.15 seconds =========================

Anything I can contribute or help with, I'm available.

sanjay900, 2024-12-24:
It feels like there's something odd going on with your pico, it's failing to copy the firmware to it, can you try a different one

XxyZwW, 2024-12-26:
You were right, I tried with another pico and yes, the programming passed, but when I tried it in operation on Xbox360 or PS3 the buttons don't work, the only thing that works is the directional pad, but it seems to interpret it as if it were the start button
I attach a short video of what it does.

https://github.com/user-attachments/assets/ad4b968f-3a3d-4cff-8905-149a3e76c1e2

Thank you and anything I can contribute or help with, I'm available.

sanjay900, 2024-12-26:
Save your current configuration using the Save Settings button, and then send the following file:
Windows: %appdata%/SantrollerConfigurator/Santroller/include/config_data.h
Linux: ~/.config/SantrollerConfigurator/Santroller/include/config_data.h
macOS: /Users/Shared/Library/Application Support/SantrollerConfigurator/Santroller/include/config_data.h

XxyZwW, 2024-12-26:
Sure, when I try to upload the file to GitHub it tells me that format is not allowed, so I compressed it in ZIP

[config_data.zip](https://github.com/user-attachments/files/18249567/config_data.zip)

sanjay900, 2024-12-26:
In santroller itself what do the inputs you can see look like?

XxyZwW, 2024-12-26:
Only "Controller Settings", "USB Host Inputs (Detected as PS3 Gamepad)" and "Pro Guitar Inputs", but the latter does not show any input

[image]
[image]
[image]
[image]

sanjay900, 2024-12-26:
oh, that explains it, it really shouldnt detect it as a ps3 gamepad!

sanjay900, 2024-12-26:
[firmware.uf2.zip](https://github.com/user-attachments/files/18251068/firmware.uf2.zip)
does this make it show up as a ps3 mustang or ps3 squire

sanjay900, 2024-12-26:
Or just try v10.1.309 when it comes out, it has the changes in it

XxyZwW, 2024-12-27:
I confirm that it is now correctly detected and works in the game, programmed with version 10.1.311, many thanks, excellent work you do

closed 2024-12-27 by sanjay900