Santroller/Santroller #230: [Issue]: Guitar Hero Drum

Psayz, 2024-10-18:
### Please describe your issue.

I'm having a problem with my drum, when I hit it once and it seems like an overstrum, it hits several times making the note hit but I lose the count, I'm using a pi pico with a 74hc4067 multiplexer, 3.3v zener diode, resistors 1 megaohm

### Version

Lastest

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-10-18:
Adjust your debounce, thats what its designed to fix

Psayz, 2024-10-18:
> Ajuste seu debounce, é para isso que ele foi projetado para consertar

I saw this on your website, but where do I find this configuration in santrollerconfig?

sanjay900, 2024-10-18:
[image]

Psayz, 2024-10-18:
There's no such thing for me

sanjay900, 2024-10-18:
click where it says controller poll settings to expand it

Psayz, 2024-10-18:
> click where it says controller poll settings to expand it

[image]

Psayz, 2024-10-18:
I reset a pi pico and work!

sanjay900, 2024-10-18:
Its because you turned on advanced mode, which changes how the debounce stuff works. You don't want advanced mode.

closed 2024-10-18 by sanjay900