Santroller/Santroller #233: [Issue]: wii config not showing all the pins

BaumLucian, 2024-11-13:
### Please describe your issue.

I just got done soldering my wires to my pi pico but I am unable to select the pins I solder them to in the config menu.
I have my SDA going to GP21 but santroller doesn't let me choose that pin.
Red is my SDA wire

[image]

[image]

### Version

10.1.274

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-11-13:
Only some pins work for certain features, if it isn't showing you can't use that pin.

BaumLucian, 2024-11-14:
okay thanks