Santroller/Santroller #234: [Issue]: Santroller fails to program ardino uno

lionclaw, 2024-11-18:
### Please describe your issue.

It happens after putting the device into dfu mode. Gives this error.

Processing arduino_uno (board: uno; platform: atmelavr; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing sukkopera/PsxNewLib @ ^0.4.0
Downloading 0% 10% 20% 30% 40% 50%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: PsxNewLib@0.4.0 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing DigitalIO
Downloading 0% 10% 20% 30%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: DigitalIO@1.0.1 has been installed!
Library Manager: Installing felis/USB-Host-Shield-20 @ ^1.6.0
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: USB-Host-Shield-20@1.6.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/uno.html
PLATFORM: Atmel AVR (5.1.0) > Arduino Uno
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 31.50KB Flash
DEBUG: Current (avr-stub) External (avr-stub, simavr)
PACKAGES: 
 - framework-arduino-avr @ 5.1.0 
 - tool-avrdude @ 1.70200.0 (7.2.0) 
 - toolchain-atmelavr @ 3.70300.220127 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 9 compatible libraries
Scanning dependencies...
Dependency Graph
|-- PsxNewLib @ 0.4.0
|-- USB-Host-Shield-20 @ 1.6.0
|-- SPI @ 1.0
|-- fxpt_math
Building in release mode
Compiling .pio\build\arduino_uno\src\avr\shared\io.cpp.o
Compiling .pio\build\arduino_uno\src\avr\shared\pins.cpp.o
Compiling .pio\build\arduino_uno\src\avr\shared\usb.cpp.o
Compiling .pio\build\arduino_uno\src\avr\uno\main\bootloader.c.o
Compiling .pio\build\arduino_uno\src\avr\uno\main\main.cpp.o
Compiling .pio\build\arduino_uno\src\shared\config\config.c.o
Compiling .pio\build\arduino_uno\src\shared\controllers\controllers.c.o
Compiling .pio\build\arduino_uno\src\shared\hid\hid.cpp.o
Compiling .pio\build\arduino_uno\src\shared\main\accel.cpp.o
Compiling .pio\build\arduino_uno\src\shared\main\i2c_shared.cpp.o
Compiling .pio\build\arduino_uno\src\shared\main\max1704x.cpp.o
Compiling .pio\build\arduino_uno\src\shared\main\mpr121.cpp.o
Compiling .pio\build\arduino_uno\src\shared\main\ps2.cpp.o
Compiling .pio\build\arduino_uno\src\shared\main\ps2_emu.cpp.o
Compiling .pio\build\arduino_uno\src\shared\main\shared_main.cpp.o
Compiling .pio\build\arduino_uno\src\shared\main\wii.cpp.o
Compiling .pio\build\arduino_uno\src\shared\main\wii_emu.cpp.o
Compiling .pio\build\arduino_uno\src\shared\usb\descriptors.cpp.o
Compiling .pio\build\arduino_uno\src\shared\usb\string_descriptors.c.o
Compiling .pio\build\arduino_uno\src\shared\usb\xbox.c.o
Compiling .pio\build\arduino_uno\src\shared\usb\xsm3\excrypt_des.c.o
Compiling .pio\build\arduino_uno\src\shared\usb\xsm3\excrypt_parve.c.o
Compiling .pio\build\arduino_uno\src\shared\usb\xsm3\excrypt_sha.c.o
src\shared\main\shared_main.cpp: In function 'int16_t handle_calibration_xbox_whammy(int16_t, uint16_t, uint32_t, int16_t, uint16_t)':
src\shared\main\shared_main.cpp:349:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (val > min) {
             ~~~~^~~~~
src\shared\main\shared_main.cpp: In function 'uint16_t handle_calibration_xbox_one_trigger(uint16_t, uint16_t, uint32_t, int16_t, uint16_t)':
src\shared\main\shared_main.cpp:375:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (val > min) {
             ~~~~^~~~~
src\shared\main\shared_main.cpp: In function 'uint16_t handle_calibration_drum(uint16_t, uint16_t, uint32_t, int16_t, uint16_t)':
[... 248 lines cut ...]
Flash: [=         ]  12.0% (used 3884 bytes from 32256 bytes)
Building .pio\build\arduino_uno_usb\firmware.hex
Configuring upload protocol...
AVAILABLE: custom
CURRENT: upload_protocol = custom
Uploading .pio\build\arduino_uno_usb\firmware.hex
dfu-programmer: no device present.
*** [upload] Error 3
========================= [FAILED] Took 13.01 seconds =========================

### Version

10.1.279

### What operating system are you using?

Windows 10

### What microcontroller are you using?

arduino uno

### How is your controller programmed?

Direct

sanjay900, 2024-11-18:
Did the drivers for the uno install properly?

lionclaw, 2024-11-18:
> Did the drivers for the uno install properly? Could you get a screenshot of the uno from device manager after it is in DFU mode?

When it isn't in DFU mode, it displays as an arduino. While in DFU, it says unknown device. 
Sancontroller tells me to install drivers every time I open the program despite clicking install. 

It may be because its an arduino clone. I've ordered a pico and will test that.

sanjay900, 2024-11-19:
the fact that it goes into dfu mode suggests it *should* work but santroller just seems to not be installing the drivers correctly for some reason

sanjay900, 2024-12-18:
Ah yeah i did trace this bug down a few versions ago, the drivers in fact were not being installed correctly and that has been resolved.

closed 2024-12-18 by sanjay900