Santroller/Santroller #236: [Issue]: Santroller Builder fails to build a Bluetooth Adapter based on Pico W

RaoulBe, 2024-11-28:
### Please describe your issue.

Trying to flash a simple BlueTooth Receiver, based on a Pico W and a Led with a resistor.
The compilation fails with the following error message:

**Observe that the standard tool can flash the same Pico with no issues.**

___Processing picow (board: rpipicow; platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/9e55f6db5c56b9867c69fe473f388bee_a4546672.zip@1.14.0; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing https://github.com/Santroller/USB_Host_Shield_2.0/archive/refs/heads/master.zip @ 1.6.2
Downloading...
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: USB-Host-Shield-20@1.6.2 has been installed!
Library Manager: Installing fortyseveneffects/MIDI Library @ 5.0.2
Downloading 0% 10% 20% 30% 40%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: MIDI Library@5.0.2 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/rpipicow.html
PLATFORM: Raspberry Pi RP2040 (1.14.0) > Pico W
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.40002.0 (4.0.2) 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-picotool-rp2040-earlephilhower @ 5.120300.240827 (12.3.0) 
 - toolchain-rp2040-earlephilhower @ 5.120300.240125 (12.3.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 67 compatible libraries
Scanning dependencies...
Dependency Graph
|-- USB-Host-Shield-20 @ 1.6.2
|-- MIDI Library @ 5.0.2
|-- SPI @ 1.0
|-- Wire @ 1.0
|-- fxpt_math
Building in release mode
Compiling .pio\build\picow\FrameworkArduinoVariant\picow_digital.cpp.o
Compiling .pio\build\picow\FrameworkArduinoVariant\picow_init.cpp.o
Compiling .pio\build\picow\FrameworkArduinoBootloader\boot2_w25q080_2_padded_checksum.S.o
Compiling .pio\build\picow\src\pico\bt.cpp.o
Compiling .pio\build\picow\src\pico\bt_rx.cpp.o
Compiling .pio\build\picow\src\pico\bt_rx_cnt.cpp.o
Compiling .pio\build\picow\src\pico\hidparser.c.o
Compiling .pio\build\picow\src\pico\io.cpp.o
Compiling .pio\build\picow\src\pico\main.cpp.o
Compiling .pio\build\picow\src\pico\midi_host.c.o
src\pico\bt_rx.cpp: In function 'void hog_start_connect()':
src\pico\bt_rx.cpp:211:18: error: could not convert 'devices[0]' from 'scan_data_t' to 'bool'
  211 |     if (devices[0]) {
      |         ~~~~~~~~~^
      |                  |
      |                  scan_data_t
*** [.pio\build\picow\src\pico\bt_rx.cpp.o] Error 1
========================= [FAILED] Took 15.49 seconds =========================__ 

### Version

10.0.183

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-11-28:
fix with v10.1.285

closed 2024-11-28 by sanjay900

RaoulBe, 2024-11-28:
Thanks

I actually found the error too 🙂. Good that it will be fixed.

/Raoul