Santroller/Santroller #239: [Issue]: the miso and mosi selections are swapped

djsedaw, 2024-12-06:
### Please describe your issue.

When selecting the pins for miso and mosi in the drop down menu, the selections for miso are the mosi pins and vice versa Versa for mosi.

### Version

Latest

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-12-07:
What exactly are you mapping a mosi and miso pin for?

djsedaw, 2024-12-07:
im connecting buttons, then joysticks via a multiplexer directly to the pico, then connecting an old ps2 gamepad cable to the mosi, miso, ack ect to the pico, then plugging into the ps2 console

sanjay900, 2024-12-07:
Then in that case its swapped on purpose. When your plugging it into the PS2, the roles are reversed thus the terminology is swapped and the names are supposed to be swapped around.

closed 2024-12-07 by sanjay900

djsedaw, 2024-12-07:
Thank you, didn't know that but now I do!