Santroller/Santroller #240: [Request]:  Multiple USB Host

kimo1arquimedes1, 2024-12-12:
### Please describe your request.

Hello, it's me again:

My request is to create a professional Rock Band drum kit using two controllers.

I have two RB1 drum brains, one for Xbox 360 and one for Wii (which is detected as PS3 in the app, though that doesn't affect its functionality). I’d like to emphasize that both controllers are detected without any issues at the same time.
When assigning an input, for example, if I set the green pad to the A button, the A button activates that mapping. However, the issue is that both Controller 1 and Controller 2 send the same information.

What I’m trying to achieve is to have one brain assigned to the drum cymbals and the other to the drum pads, but there doesn’t seem to be an option to differentiate between USB Host 1 and USB Host 2.

I know I could solve this by soldering piezoelectric sensors to the analog pins, but for now, I’d like to keep both devices intact.

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

N/A

sanjay900, 2024-12-12:
Yeah this isn't really something santroller can easily support in its current state, maybe in the future but its not something currently on the roadmap

sanjay900, 2026-04-05:
Santroller 2 was designed with this concept in mind, so its something that it will be able to do.

closed 2026-04-05 by sanjay900