Santroller/Santroller #242: [Issue]: error programming keytar pro (Raspberry Pi Pico)

kimo1arquimedes1, 2024-12-22:
### Please describe your issue.

When configuring a Raspberry Pi Pico, with the USB host and Pro keyboard combination, it gives the following error, if I remove the USB host the programming ends.

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi/archive/39b90392af50585e429941bd2561a91949a2ba46.zip@1.15.0; board: pico; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing https://github.com/Santroller/USB_Host_Shield_2.0/archive/refs/heads/master.zip @ 1.6.3
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: USB-Host-Shield-20@1.6.3 has been installed!
Library Manager: Installing fortyseveneffects/MIDI Library @ 5.0.2
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: MIDI Library@5.0.2 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.15.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, pico-debug, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.40400.0 (4.4.0) 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-openocd-rp2040-earlephilhower @ 0.0.0 
 - tool-picotool-rp2040-earlephilhower @ 5.140200.240929 (14.2.0) 
 - toolchain-rp2040-earlephilhower @ 5.140200.240929 (14.2.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
PSRAM size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 70 compatible libraries
Scanning dependencies...
Dependency Graph
|-- USB-Host-Shield-20 @ 1.6.3
|-- MIDI Library @ 5.0.2
|-- SPI @ 1.0
|-- Wire @ 1.0
|-- fxpt_math
Building in release mode
Compiling .pio\build\pico\FrameworkArduinoBootloader\boot2_w25q080_2_padded_checksum.S.o
Compiling .pio\build\pico\src\pico\bt.cpp.o
Compiling .pio\build\pico\src\pico\bt_rx.cpp.o
Compiling .pio\build\pico\src\pico\bt_rx_cnt.cpp.o
Compiling .pio\build\pico\src\pico\hidparser.c.o
Compiling .pio\build\pico\src\pico\io.cpp.o
Compiling .pio\build\pico\src\pico\main.cpp.o
Compiling .pio\build\pico\src\pico\midi_host.c.o
Compiling .pio\build\pico\src\pico\pins.cpp.o
Compiling .pio\build\pico\src\pico\xinput_device.cpp.o
Compiling .pio\build\pico\src\pico\xinput_host.cpp.o
Compiling .pio\build\pico\src\shared\config\config.c.o
Compiling .pio\build\pico\src\shared\controllers\controllers.c.o
Compiling .pio\build\pico\src\shared\hid\hid.cpp.o
Compiling .pio\build\pico\src\shared\main\accel.cpp.o
Compiling .pio\build\pico\src\shared\main\i2c_shared.cpp.o
Compiling .pio\build\pico\src\shared\main\max1704x.cpp.o
Compiling .pio\build\pico\src\shared\main\mpr121.cpp.o
Compiling .pio\build\pico\src\shared\main\ps2.cpp.o
Compiling .pio\build\pico\src\shared\main\ps2_emu.cpp.o
Compiling .pio\build\pico\src\shared\main\shared_main.cpp.o
Compiling .pio\build\pico\src\shared\main\wii.cpp.o
Compiling .pio\build\pico\src\shared\main\wii_emu.cpp.o
Compiling .pio\build\pico\src\shared\usb\descriptors.cpp.o
Compiling .pio\build\pico\src\shared\usb\string_descriptors.c.o
Compiling .pio\build\pico\src\shared\usb\xbox.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\excrypt_des.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\excrypt_parve.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\excrypt_sha.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\usbdsec.c.o
Compiling .pio\build\pico\src\shared\usb\xsm3\xsm3.c.o
[... 341 lines cut ...]
      |                                               ^~~~~
src\shared\main\shared_main.cpp:2666:13: note: in expansion of macro 'TICK_PS3'
 2666 |             TICK_PS3;
      |             ^~~~~~~~
Compiling .pio\build\pico\libe59\USB-Host-Shield-20\BTHID.cpp.o
Compiling .pio\build\pico\libe59\USB-Host-Shield-20\MiniDSP.cpp.o
Compiling .pio\build\pico\libe59\USB-Host-Shield-20\PS3BT.cpp.o
*** [.pio\build\pico\src\shared\main\shared_main.cpp.o] Error 1
========================= [FAILED] Took 33.51 seconds =========================

### Version

SantrollerConfigurator v10.1.303

### What operating system are you using?

Windows 10

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-12-22:
Was this configuration from an older version of santroller by chance? What happens if you remove the `USB Host Inputs` and then add it back again?

kimo1arquimedes1, 2024-12-22:
The configuration was done from scratch, and strangely the version I used a long time ago stopped programming the Raspberry Pi Pico, I'm not sure if any cloud file changed or something like that, because before in the old version of Santroller it didn't let me program without wifi, currently I wanted to make some modifications in 2 devices and error in both.
Currently I decided to install the latest version and I was able to program a gamepad even with the USB Host Inputs option.

The error is when wanting to combine USB Host Inputs and the Rock Band Keytar

sanjay900, 2024-12-23:
Can i get you to try v10.1.305 when it comes out? Do a fresh configuration with that one.

kimo1arquimedes1, 2024-12-23:
Perfect, now the programming finishes without problems.
Unfortunately there is no option enabled to assign keys, the console detects all the buttons, except the keys.
When I finish programming with Santroller, the keys do not appear to see if they work by default. As you can see in the image, the pad and the normal buttons appear, but the keys do not appear. I have tried with that option and there is no response on any key.
[image]

sanjay900, 2024-12-23:
Save your current configuration using the Save Settings button, and then send the following file:
Windows: %appdata%/SantrollerConfigurator/Santroller/include/config_data.h
Linux: ~/.config/SantrollerConfigurator/Santroller/include/config_data.h
macOS: /Users/Shared/Library/Application Support/SantrollerConfigurator/Santroller/include/config_data.h

sanjay900, 2024-12-23:
which keyboard are you using?

sanjay900, 2024-12-23:
Also, try v10.1.307 when it comes out, it should bring all the keys back if you do it with a fresh config and hopefully they will work

kimo1arquimedes1, 2024-12-24:
Hi, sorry for the delay.

I've already updated, but the keys are not activated according to the number of keys I press, for example key 1 activates key 2, and for example the first 5 activate several keys simultaneously. I tried directly on the console and the same problem. Here I share the configuration and a photo of the device.
[image]

kimo1arquimedes1, 2024-12-24:
[config.zip](https://github.com/user-attachments/files/18235527/config.zip)
[image]

kimo1arquimedes1, 2024-12-24:
Information for device Harmonix RB3 Keyboard for Nintendo Wii (VID=0x1BAD PID=0x3330):

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device address: 0x0003
Current configuration value: 0x01
Number of open pipes: 2

------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0110	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x08	bMaxPacketSize0   (8 bytes)
0x1BAD	idVendor
0x3330	idProduct
0x0005	bcdDevice
0x01	iManufacturer   "Licensed by Nintendo of America "
0x02	iProduct        "Harmonix RB3 Keyboard for Nintendo Wii"
0x00	iSerialNumber
0x01	bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0029	wTotalLength   (41 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0x80	bmAttributes   (Bus-powered Device)
0x32	bMaxPower      (100 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0101	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x0089	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x0A	bInterval         (10 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x01	bInterval         (1 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "Licensed by Nintendo of America "
0x02   0x0409  "Harmonix RB3 Keyboard for Nintendo Wii"

------------------------------

Connection path for device: 
Controladora de host USB genérico compatible con xHCI
Root Hub
Concentrador USB genérico
Harmonix RB3 Keyboard for Nintendo Wii (VID=0x1BAD PID=0x3330) Port: 1

Running on: Windows 10 or greater (Build Version 19045)

Brought to you by TDD v2.19.0, Dec  5 2023, 12:08:38

sanjay900, 2024-12-24:
wrong file, i need config_data.h not config.h

sanjay900, 2024-12-24:
It will only be there after you hit save settings, if it isnt there that is why

kimo1arquimedes1, 2024-12-24:
[config_data.zip](https://github.com/user-attachments/files/18235772/config_data.zip)
Sorry, I was using the portable version. I already installed it, here is the file.

kimo1arquimedes1, 2024-12-24:
[config_data.zip](https://github.com/user-attachments/files/18235778/config_data.zip)

sanjay900, 2024-12-24:
[firmware.uf2.zip](https://github.com/user-attachments/files/18235788/firmware.uf2.zip)
what happens if you load this firmware onto your pico (either use the reset to arduino button, or hold bootsel while plugging the pico into your computer, then drag the uf2 file onto it)

kimo1arquimedes1, 2024-12-24:
It works perfectly, I just need to test it on the console. The keys are detected, the touchpad works and the pedal works too (I hope). The only thing that doesn't react is the overload button. I would need to test it in the game, maybe I can check it later. 
The million dollar question, how did you fix it? hehe

sanjay900, 2024-12-24:
Let me check if i have just missed something for overdrive

sanjay900, 2024-12-24:
Should all be fixed with v10.1.308

kimo1arquimedes1, 2024-12-24:
The overdrive button works, I already tested it on the console

sanjay900, 2024-12-24:
Yeah, v10.1.308 should fix it in the config tool, and brings all the changes from that uf2 file to an actual release

kimo1arquimedes1, 2024-12-24:
Much appreciated, I hope to one day be able to use the Fender Mustang PRO-Guitar controller. (to be obtained in the future hehe)

closed 2024-12-24 by kimo1arquimedes1