Santroller/Santroller #243: [Issue]:

XxyZwW, 2024-12-23:
### Please describe your issue.

Hi, I don't know if this is a bug, I'd like to check it
I had a Host adapter programmed with an older version of the tool, I recently updated the programming with a new version of the tool to test it with an Xbox 360 wireless receiver from PC to use a guitar from this console, when testing it the guitar syncs and is detected, but after a short time it stops responding, no button works until I disconnect and reconnect the adapter, I tried it with 3 different guitars and it does the same thing, I tried to test it with older versions of the tool, but apparently they don't work anymore, because they send an error when trying to program.

### Version

10.1.305

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

other issue not controller related or I have not gotten this far

sanjay900, 2024-12-23:
try deleting `%appdata%/SantrollerConfigurator/Santroller`, `%appdata%/SantrollerConfigurator/platformio` and `%appdata%/SantrollerConfigurator/platformio.version`, and then trying an older version

sanjay900, 2024-12-24:
Can you try v10.1.308 when it comes out and see if this is fixed?

XxyZwW, 2024-12-24:
Of course, thank you very much. I will test it today as soon as I leave the office and give feedback, and Merry Christmas.

XxyZwW, 2024-12-26:
I just tested with version 10.1.308 of the tool and it works perfectly, thank you very much!

closed 2024-12-26 by sanjay900