Santroller/Santroller #244: [Issue]: Can't use MIDI drumkit

aenrione, 2024-12-24:
### Please describe your issue.

I have a Casio LD-80, which is a digital battery. This set has both MIDI IN/OUT:
[image]

Could I use this as a device for a Sancontroller? I have tried manually wiring the pads to an Arduino Pro Micro with Zener Diodes and 1Mohm resistors but seem to get a lot of internal vibration that makes it so multiple pads are pressed so I think using the builtin PCB would be a better option.

Also I have an issue when I try to flash **MIDI Input** to my pi pico on the latest version:
```bash
src/pico/main.cpp: In function 'void tick_usb()':
src/pico/main.cpp:123:5: error: 'MIDI' was not declared in this scope
  123 |     MIDI.read();
      |     ^~~~
Compiling .pio/build/pico/src/shared/controllers/controllers.c.o
*** [.pio/build/pico/src/pico/main.cpp.o] Error 1
========================== [FAILED] Took 5.46 seconds ==========================
Has error!
```

I would like Advise on how to setup this set as a Sancontroller and preferably I would like to avoid buying a MIDI to USB cable or an Audio Interface

### Version

10.1.307

### What operating system are you using?

Linux

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2024-12-24:
as of right now, only USB midi is supported.

aenrione, 2024-12-27:
Hello again @sanjay900 , i have purchased a MIDI to USB Cable like this one:
[image]

I can use it on my PC for things such as Clone Hero but haven't been able to set it up as a usb device for the Pi Pico, I know when you say USB MIDI you might be referring to the modern standard and not the 5 pin old one, but I would like some advise on what I would need to modify on the code in order to pass this device straight through the Pi Pico (i currently attached a female usb to VBUS and GPIO 2/3). Any help or suggestions are appreciated.

sanjay900, 2024-12-27:
It should just detects it when you have USB host configured

sanjay900, 2024-12-27:
https://santroller.tangentmc.net/wiring_guides/midi.html

aenrione, 2025-01-01:
@sanjay900  I don't know what I am doing wrong. I have followed the guide and all. I have soldered a Female USB to the Pi Picos VBUS, GND and some GPIO pins. When something like a controller to the pi pico it detects it as a usb host connected but when i connect the MIDI cable I get "0 Usb hosts connected". I have also tried with a USB that works with an Electric Drum Kit and still the same issue, i can't get the Pico to detect it. I think the problem may be in my USB connection, do I need something like a resistor of some sort?

sanjay900, 2025-01-01:
I have had someone say older versions seem to detect more midi devices

sanjay900, 2025-01-01:
Can you get one of these for the midi cable please?
https://www.thesycon.de/eng/usb_descriptordumper.shtml

aenrione, 2025-01-01:
This is for the MIDI IN/OUT adapter:
Information for device USB2.0-MIDI (VID=0x1A86 PID=0x752D):

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x0002
Current configuration value: 0x01
Number of open pipes: 2

------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0110	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x08	bMaxPacketSize0   (8 bytes)
0x1A86	idVendor
0x752D	idProduct
0x0254	bcdDevice
0x00	iManufacturer
0x02	iProduct        "USB2.0-MIDI"
0x00	iSerialNumber
0x01	bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0075	wTotalLength   (117 bytes)
0x02	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0x80	bmAttributes   (Bus-powered Device)
0x30	bMaxPower      (96 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x00	bNumEndPoints
0x01	bInterfaceClass      (Audio Device Class)
0x01	bInterfaceSubClass   (Audio Control Interface)
0x00	bInterfaceProtocol   (Audio Protocol undefined)
0x00	iInterface

AC Interface Header Descriptor:
------------------------------
0x09	bLength
0x24	bDescriptorType
0x01	bDescriptorSubtype
0x0100	bcdADC
0x0009	wTotalLength   (9 bytes)
0x01	bInCollection
0x01	baInterfaceNr(1)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x01	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x01	bInterfaceClass      (Audio Device Class)
0x03	bInterfaceSubClass   (MIDI Streaming Interface)
0x00	bInterfaceProtocol   (Audio Protocol undefined)
0x00	iInterface

MS Interface Header Descriptor:
------------------------------
0x07	bLength
0x24	bDescriptorType
0x01	bDescriptorSubtype
0x0100	bcdMSC
0x0051	wTotalLength   (81 bytes)

MS MIDI IN Jack Descriptor:
------------------------------
0x06	bLength
0x24	bDescriptorType
0x02	bDescriptorSubtype
0x01	bJackType
0x02	bJackID
0x00	iJack

MS MIDI IN Jack Descriptor:
------------------------------
0x06	bLength
0x24	bDescriptorType
0x02	bDescriptorSubtype
0x01	bJackType
0x03	bJackID
0x00	iJack

MS MIDI IN Jack Descriptor:
------------------------------
0x06	bLength
0x24	bDescriptorType
0x02	bDescriptorSubtype
0x02	bJackType
0x06	bJackID
0x00	iJack

MS MIDI OUT Jack Descriptor:
------------------------------
0x09	bLength
0x24	bDescriptorType
0x03	bDescriptorSubtype
0x01	bJackType
0x07	bJackID
0x01	bNrInputPins
0x06	baSourceID(1)
0x01	baSourcePin(1)
0x00	iJack

MS MIDI OUT Jack Descriptor:
------------------------------
0x09	bLength
0x24	bDescriptorType
0x03	bDescriptorSubtype
0x02	bJackType
0x04	bJackID
0x01	bNrInputPins
0x02	baSourceID(1)
0x01	baSourcePin(1)
0x00	iJack

MS MIDI OUT Jack Descriptor:
------------------------------
0x09	bLength
0x24	bDescriptorType
0x03	bDescriptorSubtype
0x02	bJackType
0x05	bJackID
0x01	bNrInputPins
0x03	baSourceID(1)
0x01	baSourcePin(1)
0x00	iJack

Endpoint Descriptor (Audio/MIDI 1.0):
------------------------------
0x09	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x02	bmAttributes      (Transfer: Bulk / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (32 bytes)
0x00	bInterval         
0x00	bRefresh
0x00	bSynchAddress

MS Bulk Data Endpoint Descriptor:
------------------------------
0x06	bLength
0x25	bDescriptorType
0x01	bDescriptorSubtype
0x02	bNumEmbMIDIJack
0x02	baAssocJackID(1)
0x03	baAssocJackID(2)

Endpoint Descriptor (Audio/MIDI 1.0):
------------------------------
0x09	bLength
0x05	bDescriptorType
0x82	bEndpointAddress  (IN endpoint 2)
0x02	bmAttributes      (Transfer: Bulk / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (32 bytes)
0x00	bInterval         
0x00	bRefresh
0x00	bSynchAddress

MS Bulk Data Endpoint Descriptor:
------------------------------
0x05	bLength
0x25	bDescriptorType
0x01	bDescriptorSubtype
0x01	bNumEmbMIDIJack
0x07	baAssocJackID(1)

Microsoft OS Descriptor:
------------------------------
0x00	bLength
0x00	bDescriptorType
Hex dump: 

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x02   0x0409  "USB2.0-MIDI"

------------------------------

Connection path for device: 
USB xHCI Compliant Host Controller
Root Hub
USB2.0-MIDI (VID=0x1A86 PID=0x752D) Port: 7

Running on: Windows 10 or greater (Build Version 19045)

Brought to you by TDD v2.19.0, Dec  5 2023, 12:08:38

aenrione, 2025-01-01:
I am trying to use older versions but for usb host i am getting the same errors as [This Issue](https://github.com/Santroller/Santroller/issues/245)
------------------------------

sanjay900, 2025-01-01:
try v10.1.302

aenrione, 2025-01-02:
I have tried that specific version and I have a couple of observations:
- When installed as Windows setup, it now detects my E-kit through usb-b but it does not detect the Usb midi in/out

Disregarding the undetected one. Is there a way I could configure which MIDI notes trigger which input? For example by default the kick pedal is input 35, but I think my Kit has another mapping since in Clone Hero is different, so now I don't know 
how to map the inputs.

sanjay900, 2025-01-02:
Yes, you just hit add setting, pick say kick, then set the input to midi input and pick the note in question

aenrione, 2025-01-02:
Thanks i can finally use it correctly, sorry for all the questions, you've been very helpful!

closed 2025-01-02 by aenrione

sanjay900, 2025-01-02:
In theory all those changes are now in the latest release as well

aenrione, 2025-01-14:
@sanjay900 i ran into a weird issue. Using my midi drum kit i have mapped some buttons to the pi pico. In the application the inputs are detected correctly and i can see when i press each note on the kit, but if i connect the santroller to my xbox 360 the input of the drums only works for a few seconds before not being recognized anymore. Has this issue ever happened before to someone or any idea what could be happening? I have an rghed console with the usbsecpatch running in dash launch.

sanjay900, 2025-01-14:
Try a powered USB hub maybe, I know the consoles have issues providing power sometimes