Santroller/Santroller #247: [Request]: Support for Fender Telecaster (7-wire passthrough fret boards)

SnubbleJr, 2025-01-02:
### Please describe your request.

It doesn't look like there's a way tto hook up 7-wire RB guitar necks. Here's a photo of the solo board in my Telecaster. The main board is what you'd expect.

TBH I'm not sure how it's supposed to work, so I might be missing a setting.

[image]

### What operating system are you using?

Windows 10

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2025-01-02:
Just run new wires yourself, then you can run each switch individually

sanjay900, 2025-01-02:
Other option otherwise is to set the solo frets as "shortcuts" that require the extra solo modifier as an extra input (that's all the extra wire is, it sends an extra button when a solo fret is pressed)

SnubbleJr, 2025-01-02:
Aha! The shortcuts feature! Thaank you so much for telling me about that! it works great thanks!

closed 2025-01-02 by SnubbleJr