Santroller/Santroller #248: [Issue]: auto-activated tilt on gh:wor (rpcs3)

Derpytron84, 2025-01-03:
### Please describe your issue.

i've been playing warriors of rock a lot on rpcs3 recently, but whenever the star power tilt is on, it automatically activates without me doing anything. seeing as it's internally mapped to _**rs axis y+**_, i tried messing with the whammy controls to get the stick input to center, and while that fixed the problem i had with the whammy bar, the star power tilt always keeps going off by itself, no matter which way i tilt the guitar or which way it sits.

i've tried everything i could to fix this: inverting the controls on the emulator, messing with deadzones, calibrating the tilt with joy.cpl, mapping the tilt pin to analog controls, and even inverting it to see if anything different happened.
nothing on the emulator seemed to change anything, and i couldn't find anything on the configurator to fix it either. if there's anything else i can do to see if this fixes the issue, please let me know. i don't want to develop any blisters by tapping the tiny back button on my controller every time i activate star power

ps: i'm using my xplorer which i modded with a solderless kit from retrocult. it's being read as an xinput controller on my computer, and is set as a guitar hero controller in the emulator.

### Version

10.1.316

### What operating system are you using?

Windows 11

### What microcontroller are you using?

pi pico

### How is your controller programmed?

Direct

sanjay900, 2025-01-03:
I think rpcs3 has a function for that where you can right click on the axis to make an input full range, they got a lot of complaints about this because actual hardware does this

sanjay900, 2025-01-03:
Your easiest method would just be to map tilt to the back button, I don't actually think you can get tilt any other way since the GH guitars put tilt into the motion controls area of the gamepad report, which is not something easily mapped with RPCS3

sanjay900, 2025-01-03:
https://github.com/RPCS3/rpcs3/issues/9149

closed 2025-01-03 by sanjay900