Santroller/Santroller #251: [Issue]: Arduino not able to be configured

g0ldm1n3r, 2025-01-07:
### Please describe your issue.

Any help would be appreciated

Info:
OS Version: 10.0.19045.0
OS Platform: WIN32NT
OS Service Pack:
OS Version String: Microsoft Windows NT 10.0.19045.0

Device Type: GuitarHeroGuitar
LED Type: None

 Microcontroller: Arduino Leonardo 
Microcontroller Frequency: 16000mhz

**Error log:**

Processing microdetect (platform: atmelavr; board: micro; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/micro.html
PLATFORM: Atmel AVR (5.1.0) > Arduino Micro
HARDWARE: ATMEGA32U4 16MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) External (simavr)
PACKAGES: 
 - framework-arduino-avr @ 5.1.0 
 - tool-avrdude @ 1.70200.0 (7.2.0) 
 - toolchain-atmelavr @ 3.70300.220127 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- LUFA
Building in release mode
mytarget_callback2(["micro_clean_existing"], [])
Waiting for bootloader device
Waiting for the new upload port...
PORT: COM4
"C:\Users\thegoldminer9\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages/tool-avrdude/avrdude" -C "C:\Users\thegoldminer9\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages/tool-avrdude/avrdude.conf" -p atmega32u4 -P COM4 -c avr109 -e
avrdude: AVR device initialized and ready to accept instructions
ignature = 0x1e9587 (probably m32u4)
avrdude: erasing chip
avrdude done.  Thank you.
Processing arduino_leonardo_16 (board: leonardo; platform: atmelavr; framework: arduino)
--------------------------------------------------------------------------------
Library Manager: Installing felis/USB-Host-Shield-20 @ ^1.6.0
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: USB-Host-Shield-20@1.6.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/leonardo.html
PLATFORM: Atmel AVR (5.1.0) > Arduino Leonardo
HARDWARE: ATMEGA32U4 16MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) External (simavr)
PACKAGES: 
 - framework-arduino-avr @ 5.1.0 
 - tool-avrdude @ 1.70200.0 (7.2.0) 
 - toolchain-atmelavr @ 3.70300.220127 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- USB-Host-Shield-20 @ 1.6.0
|-- LUFA
|-- fxpt_math
Building in release mode
Compiling .pio\build\arduino_leonardo_16\src\avr\lufa\LUFA.c.o
Compiling .pio\build\arduino_leonardo_16\src\avr\lufa\reboot.c.o
Compiling .pio\build\arduino_leonardo_16\src\avr\micro\main\main.cpp.o
Compiling .pio\build\arduino_leonardo_16\src\avr\micro\shared\bootloader.c.o
Compiling .pio\build\arduino_leonardo_16\src\avr\shared\io.cpp.o
Compiling .pio\build\arduino_leonardo_16\src\avr\shared\pins.cpp.o
Compiling .pio\build\arduino_leonardo_16\src\avr\shared\usb.cpp.o
Compiling .pio\build\arduino_leonardo_16\src\shared\config\config.c.o
Compiling .pio\build\arduino_leonardo_16\src\shared\controllers\controllers.c.o
In file included from C:\Users\thegoldminer9\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages\framework-arduino-avr\libraries\LUFA/LUFA/LUFA/Drivers/USB/Core/ConfigDescriptors.h:56:0,
                 from C:\Users\thegoldminer9\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages\framework-arduino-avr\libraries\LUFA/LUFA/LUFA/Drivers/USB/USB.h:389,
                 from C:\Users\thegoldminer9\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages\framework-arduino-avr\libraries\LUFA/LUFA.c.inc:43,
                 from src\avr\lufa\LUFA.c:3:
C:\Users\thegoldminer9\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages\framework-arduino-avr\libraries\LUFA/LUFA/LUFA/Drivers/USB/Core/HostStandardReq.h:281:20: warning: 'USB_Host_SendControlRequest_PRV' declared 'static' but never defined [-Wunused-function]
     static uint8_t USB_Host_SendControlRequest_PRV(void* const BufferPtr);
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\thegoldminer9\AppData\Local\SantrollerConfigurator\current\Binaries\platformio\packages\framework-arduino-avr\libraries\LUFA/LUFA/LUFA/Drivers/USB/Core/HostStandardReq.h:282:20: warning: 'USB_Host_WaitForIOS' declared 'static' but never defined [-Wunused-function]
     static uint8_t USB_Host_WaitForIOS(const uint8_t WaitType);
                    ^~~~~~~~~~~~~~~~~~~
Compiling .pio\build\arduino_leonardo_16\src\shared\hid\hid.cpp.o
Compiling .pio\build\arduino_leonardo_16\src\shared\main\accel.cpp.o
Compiling .pio\build\arduino_leonardo_16\src\shared\main\i2c_shared.cpp.o
Compiling .pio\build\arduino_leonardo_16\src\shared\main\max1704x.cpp.o
Compiling .pio\build\arduino_leonardo_16\src\shared\main\mpr121.cpp.o
Compiling .pio\build\arduino_leonardo_16\src\shared\main\ps2.cpp.o
src\avr\micro\main\main.cpp: In function 'void EVENT_USB_Device_ControlRequest()':
src\avr\micro\main\main.cpp:96:191: error: too few arguments to function 'uint16_t controlRequest(uint8_t, uint8_t, uint16_t, uint16_t, uint16_t, uint8_t*, bool*)'
             uint16_t len = controlRequest(USB_ControlRequest.bmRequestType, USB_ControlRequest.bRequest, USB_ControlRequest.wValue, USB_ControlRequest.wIndex, USB_ControlRequest.wLength, buf);
                                                                                                                                                                                               ^
In file included from src\avr\micro\main\main.cpp:22:0:
include/descriptors.h:124:10: note: declared here
 uint16_t controlRequest(const uint8_t requestType, const uint8_t request, const uint16_t wValue, const uint16_t wIndex, const uint16_t wLength, uint8_t* requestBuffer, bool* status);
          ^~~~~~~~~~~~~~
src\avr\micro\main\main.cpp:103:176: error: too few arguments to function 'uint16_t controlRequest(uint8_t, uint8_t, uint16_t, uint16_t, uint16_t, uint8_t*, bool*)'
             controlRequest(USB_ControlRequest.bmRequestType, USB_ControlRequest.bRequest, USB_ControlRequest.wValue, USB_ControlRequest.wIndex, USB_ControlRequest.wLength, buf);
                                                                                                                                                                                ^
In file included from src\avr\micro\main\main.cpp:22:0:
include/descriptors.h:124:10: note: declared here
 uint16_t controlRequest(const uint8_t requestType, const uint8_t request, const uint16_t wValue, const uint16_t wIndex, const uint16_t wLength, uint8_t* requestBuffer, bool* status);
          ^~~~~~~~~~~~~~
Compiling .pio\build\arduino_leonardo_16\src\shared\main\ps2_emu.cpp.o
*** [.pio\build\arduino_leonardo_16\src\avr\micro\main\main.cpp.o] Error 1
========================== [FAILED] Took 5.91 seconds ==========================

### Version

v10.1.323

### What operating system are you using?

Windows 10

### What microcontroller are you using?

arduino pro micro 5V

### How is your controller programmed?

Direct

sanjay900, 2025-01-07:
Damn I missed one, I'll fix it soon

sanjay900, 2025-01-07:
Use v10.1.316 for now though it should work

g0ldm1n3r, 2025-01-07:
Got it, thanks for the help.

sanjay900, 2025-01-07:
this is fixed with v10.1.324 as well.

closed 2025-01-07 by sanjay900