Santroller/Santroller #253: [Request]: Better Controller name

zenaro147, 2025-01-14:
### Please describe your request.

I believe it's a simple request.

I currently use the Santroller with Guitar Hero World Tour Definitive Edition, and I use all 3 instruments (2 guitars and 1 drums) to play with my family and friends.

But I noticed that is a little difficulty to setup all controllers, since all 3 controllers name is showing as "Santroller" only.
[image]

Is it possible to add some kind of "second name" like "Santroller Guitar [random ID or something unique]", to help to identify the controllers? This will helpful if the player don't want to play with all instruments all the time and, at least on GHWT, if I play only with drums, the game can randomly bind my Drums to the guitar slot, like the image below
[image]

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

N/A

sanjay900, 2025-01-16:
Its not quite as simple as youd hope, especially on windows.
Windows does some caching where if a bunch of XInput devices have the same VID and PID, it shares the device name across all of them

zenaro147, 2025-01-16:
> Its not quite as simple as youd hope, especially on windows. Windows does some caching where if a bunch of XInput devices have the same VID and PID, it shares the device name across all of them

Oh I also forgot to mentined that I use HID, not xinput.
If you want I can share any device info.

But I know that xinput have some specs, but should be very handy to configure and identify devices using multiple adapters 😀

sanjay900, 2025-01-16:
I can probably do it for HID

sanjay900, 2025-01-16:
It's interesting because I know rewired (which is what ch uses internally) uses the serial number as part of it's algorithm for pairing controllers, so I don't quite know why it's not working there.

sanjay900, 2025-01-16:
What type of pico are you using?

zenaro147, 2025-01-16:
> What type of pico are you using?

All my boards are generic Pico boards (not the W ones) all boards are connected using USB cable on my PC

zenaro147, 2025-01-16:
> It's interesting because I know rewired (which is what ch uses internally) uses the serial number as part of it's algorithm for pairing controllers, so I don't quite know why it's not working there.

Yeah.. I mean.. The binds still there.. but every time I need to reassign the controllers for each player (since my PC is also my Arcade, I already have 2 arcade sticks and 2 vjoys for my lightguns always connected, but CH always lost the player controller assign and reset to a default one, but the key binds still working fine)

zenaro147, 2025-01-16:
@sanjay900 according to my topic on WTDE on discord, they say that the game uses the controller name to bind the commands...

So probably if each controller have a unique name (or santroller+Type, like Santroller Guitar, Santroller Drums) It should fix the WT issue