Santroller/Santroller #255: [Request]: Skylanders base portal emulation

sonik-br, 2025-01-16:
### Please describe your request.

I see that Skylanders portal can now be used as auth device and this is amazing!
It's possible to also make the portal from one console work on another? I'm not sure but if they use the same protocol with just different IDs, it might be possible.
Using the portal's leds as rock band leds would also be a cool feature :)

### What operating system are you using?

Windows 11

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2025-01-16:
That's something I will probably do in the future but the portals work so different to everything else that I probably won't do it for a while

sanjay900, 2025-02-17:
https://github.com/sanjay900/portal_of_flipper.git
eventually this stuff will flow back to santroller, but it does look like itll be possible to adapt portals across consoles

sonik-br, 2025-02-18:
Amazing!

I do have a x360 trap team portal and I've been doing some experiments with it. The protocol is the same indeed.
I could not get the audio part of it working. Probably it uses the dedicated audio interface.

sanjay900, 2025-02-18:
It doesn't, it uses a similar audio interface to hid, only it uses a different header and 32 byte packets