Santroller/Santroller #260: [Request]: Drawing tablet emulation

TheEssem, 2025-01-20:
### Please describe your request.

Some drawing tablet-like devices were released for game consoles back in the day, such as the Drawsome Tablet (for Wii) and uDraw GameTablet (for Wii, PS3, and Xbox 360). Considering that Santroller supports both of the aforementioned devices as input, I wonder if it would be possible to expose them as regular drawing tablets on the host side (potentially using the USB HID digitizers usage page [0x0D]). This could be useful for drawing applications such as Krita and Clip Studio Paint, as well as rhythm games utilizing touchscreen/touchpad/pen controls.

### What operating system are you using?

Linux

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

Direct

sanjay900, 2025-01-20:
Possibly in the future, i think for now the best you could do is probably map it to a mouse since that is something we already emulate

sanjay900, 2025-01-20:
[image]
Mouse emulation + absolute movement should be good enough for a drawing tablet

TheEssem, 2025-01-20:
> [image] Mouse emulation + absolute movement should be good enough for a drawing tablet

My main concern about this approach is the lack of pressure sensitivity support; the pen on the uDraw GameTablet in particular sends tip pressure data, and as far as I can tell it is not currently possible to expose this data properly to applications with the mouse approach.

sanjay900, 2025-01-20:
Yeah, thats a later thing to look into

ioletsgo, 2025-01-20:
Hi, I also have a uDraw tablet, and a santroller adapter (rcm v3), and I am also interested in this topic, just as a silly curiosity.

For some reason it seems like my uDraw tablet doesn't seem to function? It registers clicks but it doesn't register the actual tablet in use. idk if that's just a me thing or if there's another way to test if its actually functioning as intended

sanjay900, 2025-01-21:
I don't actually have one of these tablets so it might be broken, but also is the pen position actually mapped to something?