Santroller/Santroller #264: Error building: src\pico\sleep.c:34:10: fatal error: hardware/powman.h: No such file or directory

krin-san, 2025-01-25:
OS Version: 10.0.19045.0
OS Platform: Win32NT
OS Service Pack: 
OS VersionString: Microsoft Windows NT 10.0.19045.0

Device Type: GuitarHeroGuitar
Led Type: None

Microcontroller Type: Raspberry PI Pico 2
Microcontroller Frequency: 125000mhz

Compilation Log: https://hastebin.com/share/onenikomij.less

---

Error raised during Configuration phase.

Seems to be caused by e96e96ab1ee4dceaf1f48636067940d88cfdc59b where `#include "hardware/powman.h"` has first been introduced.

Relative paths:
```
SantrollerConfigurator-win-Portable\current\Binaries\Santroller\src\pico\sleep.c
SantrollerConfigurator-win-Portable\current\Binaries\platformio\packages\framework-arduinopico\pico-sdk\src\rp2_common\hardware_powman\include\hardware\powman.h
```

sanjay900, 2025-01-25:
Guess there's some import missing there for the pico 2, I'll get mine out and have a look

sanjay900, 2025-01-25:
Fixed with v10.1.348

closed 2025-01-25 by sanjay900