Santroller/Santroller #265: [Request]: Option to disable HID mode completly, so only Xinput mode is used

guzzard, 2025-01-26:
### Please describe your request.

Hi! First of all thanks for an awesome project and all the hard work you and others have put into this!

I want to use Santroller for a PS2 to Xinput/XBOX 360 gamepad emulator. On Windows this works great, and the PS2 controller appears as an Xinput device with all the normal buttons, with all mapping correct and working great. However on my LG TV running WebOS and my Linux computer the controller appears as a HID device as expected.

I was wondering if it would be possible to add an option to disable HID mode completely so the controller always appears as a Xinput device? Or perhaps another drop-down "Other OS controller mode" where you can choose if the device should appear as HID or Xinput?

### What operating system are you using?

Other

### os-other

WebOS (LG TV) and Linux

### What microcontroller are you making a request for?

pi pico

### How will the microcontroller be programed for this request?

PS2

sanjay900, 2025-01-26:
You'd just have to use a 360 console mode binding, it doesn't really work to force xinput mode on things.

It shouldn't really matter on Linux, since most games are looking at the hid layout and we do mostly mirror xinput there, which is why stuff does just work in wine.

sanjay900, 2025-01-26:
Is there a specific problem you are trying to resolve, like some axis being flipped in games or something?

sanjay900, 2025-01-26:
all the buttons are wrong arent they? thats the problem your trying to solve right?

sanjay900, 2025-01-26:
v10.2.0 will make the mappings of HID mode actually be correct, with that change my gamepad just worked in games on linux, and it would not surprise me if it also fixes your webos tv as well

guzzard, 2025-01-26:
Hi Sanjay!

I have two use cases that I wanted to use Santroller for:

1. Use a PS2 controller connected to an LG TV running [Moonlight](https://github.com/mariotaku/moonlight-tv) to connect to a Windows PC running [Sunshine](https://github.com/LizardByte/Sunshine) to stream games to the TV. Sunshine tries to map HID devices to an emulated PS4 controller, and Xinput devices to an Xbox controller When I tested yesterday with Santroller (PS2 adapter to gamepad) the controls was all messed up both in the LG TV Moonlight interface, as well as in Windows where the controls are forwarded to. The LG TV also has much better compatibility with standard Xinput controls such as XBOX 360 controllers, XBOX One controllers etc. With HID controllers it's a bit hit or miss.. sometimes it works great, sometimes the controls are all weird.

2. Use a PS2 controller connected to my main Linux computer is the second use case. The controls worked, and did not look into if it was following a standard mapping or not since you normally can map HID controllers in game settings etc.. but would be nice to be able to use Xinput instead, since most games have support for it without needing to configure anything.. I often use a wired XBOX 360 controller that works great and always "just works".. would be awesome if it was possible to force spoof the controller so it appears as if it was just a genuine XBOX 360 controller? As I understand that is possible, and how it's done when connected to a console?

Thanks again for all the work on this project! I will test v10.2.0 and report back. Perhaps everything above is working great already with that version.

guzzard, 2025-01-26:
> v10.2.0 will make the mappings of HID mode match what Linux wants. With that change my gamepad just worked in games on linux, and it should fix anything based on the linux kernel, webOS included.

Here are some results after testing the new version.

**Linux**
Mostly okay, but a bit of mixed result.
- In [jstest-gtk](https://github.com/Grumbel/jstest-gtk) all the buttons and axis registered input, but button 2 was unused (perhaps as it should be?) (X=0, O=1, Square=3, Triangle=4). The Y-axis for both thumb-sticks were inverted.
- In [Gamepad Tester](https://hardwaretester.com/gamepad) Y-axis for both thumb-sticks were also inverted, and mapping as a bit different with Square and Triangle switched (X=B0, O=B1, Square=B3, Triangle=B2).
- In Steam (Settings > Controller > Test Device Inputs) all buttons mapped correctly, but Y-axis for both thumb-sticks were inverted.

**WebOS in menus etc**
- Controller mapping still totally messed up.

**Windows (LG TV HID device > Moonlight > Sunshine (emulated PS DS4 controller since HID device) > Windows)**
- Controller mapping still totally messed up.

Would be very interesting to just try the same tests with Santroller spoofing a XBOX 360 controller, or with forced Xinput. I think especially WebOS and also passing the controller through Moonlight/Sunshine would work much better.

sanjay900, 2025-01-26:
Yeah, spoofing IDs would work but you can probably understand why that's not something we do by default.
You could probably even get away with just using a PS3 console mode binding, that would likely also work and we do spoof IDs in that scenario.

Axis being flipped is something I can fix.

guzzard, 2025-01-26:
Sorry I did not fully understand how to use mode binding. I did some reading and have good news! I got it working so that the LG TV now recognizes Santroller as a Xinput device using the Xbox 360 Console Mode Binding.

There was one issue but I managed to work around it, and it's related to the pressure sensitive buttons on the PS2 DS2 controllers. When I first used X (cross) as Input (to activate the console binding mode), that button then required to be pressed fully down hard to register input. I also tested O (circle) and it was the same. I then changed to the Select button instead which is a digital button without the pressure sensitivity, and now it works great. The Select button works as expected with Xbox 360 Console Mode Binding activated, and no need to press very hard. All other buttons also work as expected and Sunshine detects the controller as an original XBOX 360 controller when passed through Moonlight on the TV. Awesome! Thanks a lot for your help and feedback!

(Less important stuff below..)
On Linux without Xbox 360 Console Mode Binding: both HID device and Xbox 360 controller is detected and register inputs (js0, js1), but all inputs are registered from the HID device.

On Linux with Xbox 360 Console Mode Binding active: first both HID device and Xbox 360 controller is detected as before, but then Santroller reboots to activate the Xbox 360 Console Mode Binding. When that happens Linux fails to register any input devices. Log looks like this:

sanjay900, 2025-01-26:
Do you have steam open by chance?

guzzard, 2025-01-26:
> Do you have steam open by chance? That Xbox 360 controller showing up looks an awful lot like steam creating its own 360 pad, and not like santroller doing that. That's why it has no inputs either. How old is your Linux kernel?

Kernel: [6.11.10-1-ck](https://aur.archlinux.org/packages/linux-ck)

Sorry, I'm an idiot..

1. Yes, you are correct. It was Steam adding that XBOX 360 controller. I did not know Steam do that..
2. The reason why it didn't work with Xbox 360 Console Mode Binding enabled was because I had blacklisted xpad. I used xboxdrv before. After enabling xpad Santroller with Xbox 360 Console Mode Binding enabled worked fine!

Log:
```
[  265.609603] usb 1-6.3.2: new full-speed USB device number 16 using xhci_hcd
[  265.714588] usb 1-6.3.2: New USB device found, idVendor=1209, idProduct=2882, bcdDevice= 1.00
[  265.714604] usb 1-6.3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  265.714611] usb 1-6.3.2: Product: Santroller
[  265.714617] usb 1-6.3.2: Manufacturer: sanjay900
[  265.714622] usb 1-6.3.2: SerialNumber: E46274A15F3A3D243011
[  265.725593] input: sanjay900 Santroller as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.3/1-6.3.2/1-6.3.2:1.0/0003:1209:2882.000E/input/input20
[  265.726037] hid-generic 0003:1209:2882.000E: input,hidraw13: USB HID v1.01 Gamepad [sanjay900 Santroller] on usb-0000:00:14.0-6.3.2/input0
[  266.436389] usb 1-6.3.2: USB disconnect, device number 16
[  266.659609] usb 1-6.3.2: new full-speed USB device number 17 using xhci_hcd
[  266.764628] usb 1-6.3.2: New USB device found, idVendor=1209, idProduct=2882, bcdDevice= 1.00
[  266.764643] usb 1-6.3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  266.764650] usb 1-6.3.2: Product: Santroller
[  266.764656] usb 1-6.3.2: Manufacturer: sanjay900
[  266.764661] usb 1-6.3.2: SerialNumber: E46274A15F3A3D243111
[  266.783725] input: Ardwiino Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.3/1-6.3.2/1-6.3.2:1.0/input/input21
``` 

So it's all great! Everything is working that I wanted! Super happy! Thanks again!

closed 2025-01-26 by guzzard

sanjay900, 2025-01-27:
BTW i have flippped the y axis now in v10.2.3 with hid mode, so that should work with more things without you needing to use the xbox 360 console mode binding

guzzard, 2025-01-28:
> BTW i have fixed the inverted the y axis now in v10.2.3 with hid mode, so that should work with more things without you needing to use the xbox 360 console mode binding

Thank you! Working great!