Santroller/Santroller #268: [Issue]/[Request]: Support for SK6812 LED

zampierilucas, 2025-01-28:
### Please describe your issue.

Hello,

I’m not sure if this should be classified as an issue or a feature request, but I’ve been trying to use the SK6812 Mini-E[1], and I’ve noticed that the color mapping doesn’t work correctly—it appears inconsistent and somewhat glitchy.

I’ve successfully used this specific LED with the default WS2812B library in the past without any issues, so I initially expected it to function seamlessly here as well, given that this project supports it.

For reference, I’ve used the SK6812 Mini-E with QMK's WS2812B implementation without any problems.

Here’s the datasheet for the LED:
https://cdn-shop.adafruit.com/product-files/4960/4960_SK6812MINI-E_REV02_EN.pdf

Any guidance or suggestions would be greatly appreciated! Thank you!

### Version

10.2.4

### What operating system are you using?

Linux

### What microcontroller are you using?

arduino pro micro 5V

### How is your controller programmed?

Direct

sanjay900, 2025-01-29:
Yeah one of these days i need to redo that stuff, just havent had the time

sanjay900, 2025-01-29:
hmm, it seems to be working perfectly fine on my pi pico, ill have to get out a pro micro and test with that as well

sanjay900, 2025-01-29:
damn actually i think this SPI code wont work on the arduinos, and thus i probably just outright won't support the ws2812 on them.

closed 2025-01-29 by sanjay900

zampierilucas, 2025-01-29:
@sanjay900 Thanks for the quick response, I'll replace the arduino with a pi pico then.

sanjay900, 2025-01-29:
Yeah, that is certainly the best way to go, picos are just much better at everything.